pagermon/pagermon #169: Capcode Refresh Undefined Error

DanrwAU, 2018-12-31:
Getting an undefined error when refreshing capcodes on the big instance (1900 capcodes)

`Error Refreshing Aliases: undefined`

No error is logged to the logs even with loglevel of debug. 
Any thoughts on some extra debugging i can throw in to figure out why it's throwing? It does *seem* to be refreshing correctly.

DanrwAU, 2018-12-31:
Running the refresh from the SQLITE3 console fixed the issue and it can now refresh successfully from the web *sigh*

davidmckenzie, 2018-12-31:
Voodoo... how long does it take to get that error? Could be http timeout of some kind?

DanrwAU, 2018-12-31:
60ish seconds ... I thought the same but couldnt find a timeout value anywhere.
More digging required

DanrwAU, 2018-12-31:
Found the problem, NGINX was timing out when proxying the connection across. 

For anyone who comes across it in the future, you need to set this in the location block - obviously times can be adjusted to suit. 

proxy_connect_timeout       605;
proxy_send_timeout          605;
proxy_read_timeout          605;
send_timeout                605;
keepalive_timeout           605;

closed 2018-12-31 by DanrwAU