pagermon/pagermon #170: Telegram and Pushover

fletch19, 2019-01-01:
Hi,
I have managed to get the app up and running on a Raspberry Pi, however, both Telegram and Pushover are not receiving messages.
I have checked the logs in  /home/pi/SDR/pagermon/server/logs node-app.stdout.log
and the following is there:-

[pmon]  2019-01-01 14:01:81  [34mdebug[39m : ======================
[pmon]  2019-01-01 14:01:81  [34mdebug[39m : ======================
[pmon]  2019-01-01 14:01:82  [34mdebug[39m : plugin: Telegram
[pmon]  2019-01-01 14:01:82  [34mdebug[39m : Plugin does not run in this scope
[pmon]  2019-01-01 14:01:82  [34mdebug[39m : ======================
[pmon]  2019-01-01 14:01:83  [34mdebug[39m : plugin: Pushover
[pmon]  2019-01-01 14:01:83  [34mdebug[39m : Plugin does not run in this scope

I have tried changing the scope in Telegram.json, but the error message remains the same. Has anyone any ideas?

davidmckenzie, 2019-01-01:
It's definitely meant to be an 'after' plugin :) Are there more logs after that bit?

fletch19, 2019-01-02:
Hi, 
Here are some more logs.

 pluginData: {} }
[pmon]  2019-01-01 19:01:35  [34mdebug[39m : plugins object
[pmon]  2019-01-01 19:01:35  [34mdebug[39m : { Telegram: 
   { enable: true,
     teleAPIKEY: 'xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx' },
  Pushover: { enable: true, pushAPIKEY: 'xxxxxxxxxxxxxxxxxxxxxxxx' } }
[pmon]  2019-01-01 19:01:35  [34mdebug[39m : ======================
[pmon]  2019-01-01 19:01:35  [34mdebug[39m : ======================
[pmon]  2019-01-01 19:01:35  [34mdebug[39m : plugin: Telegram
[pmon]  2019-01-01 19:01:35  [34mdebug[39m : Plugin does not run in this scope
[pmon]  2019-01-01 19:01:35  [34mdebug[39m : ======================
[pmon]  2019-01-01 19:01:35  [34mdebug[39m : plugin: Pushover
[pmon]  2019-01-01 19:01:36  [34mdebug[39m : Plugin does not run in this scope
[pmon]  2019-01-01 19:01:36  [34mdebug[39m : { address: '000000',
  message: ',etc,END,,',
  datetime: '12345678',
  source: 'pager',
  pluginData: {} }
[pmon]  2019-01-01 19:01:36  [34mdebug[39m : beforeMessage done
[pmon]  2019-01-01 19:01:84  [34mdebug[39m : afterMessage start
[pmon]  2019-01-01 19:01:84  [34mdebug[39m : ======================
[pmon]  2019-01-01 19:01:84  [34mdebug[39m : trigger: message scope: after
[pmon]  2019-01-01 19:01:84  [34mdebug[39m : ======================
[pmon]  2019-01-01 19:01:84  [34mdebug[39m : data object
[pmon]  2019-01-01 19:01:84  [34mdebug[39m : { id: 357,
  address: '000000',
  message: ',etc,END,,',
  source: 'pager',
  timestamp: 
  alias_id: null,
  alias: null,
  agency: null,
  icon: null,
  color: null,
  ignore: null,
  aliasMatch: null,
  pluginconf: {},
  pluginData: {} }
[pmon]  2019-01-01 19:01:85  [34mdebug[39m : plugins object
[pmon]  2019-01-01 19:01:85  [34mdebug[39m : { Telegram: 
   { enable: true,
     teleAPIKEY: 'xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx' },
  Pushover: { enable: true, pushAPIKEY: 'xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx' } }
[pmon]  2019-01-01 19:01:85  [34mdebug[39m : ======================
[pmon]  2019-01-01 19:01:85  [34mdebug[39m : ======================
[pmon]  2019-01-01 19:01:85  [34mdebug[39m : plugin: Telegram
[pmon]  2019-01-01 19:01:85  [34mdebug[39m : RUNNING PLUGIN!
[pmon]  2019-01-01 19:01:85  [34mdebug[39m : ======================
[pmon]  2019-01-01 19:01:85  [34mdebug[39m : plugin: Pushover
[pmon]  2019-01-01 19:01:86  [34mdebug[39m : RUNNING PLUGIN!
[pmon]  2019-01-01 19:01:86  [34mdebug[39m :{ id: 357,
  address: '000000',
  message: 'etc,,END,,',
  source: 'pager',
  timestamp: ,

  alias_id: null,
  alias: null,
  agency: null,
  icon: null,
  color: null,
  ignore: null,
  aliasMatch: null,
  pluginconf: {},
  pluginData: {} }
[pmon]  2019-01-01 19:01:86  [34mdebug[39m : afterMessage done
::ffff:127.0.0.1 - - [01/Jan/2019:19:53:46 +0000] "POST /api/messages HTTP/1.1" 200 3 "-" "-"
::ffff:192.168.1.125 - - [01/Jan/2019:20:15:55 +0000] "GET / HTTP/1.1" 200 - "http://192.168.1.88:3000/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/64.0.3282.140 Safari/537.36 Edge/17.17134"
::ffff:192.168.1.125 - - [01/Jan/2019:20:15:59 +0000] "GET /stylesheets/style.css HTTP/1.1" 200 - "http://192.168.1.88:3000/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/64.0.3282.140 Safari/537.36 Edge/17.17134"
init: 1.755ms
sql: 8.185ms
send: 6.934ms

marshyonline, 2019-01-02:
Have you got a alais set for this capcode?

davidmckenzie, 2019-01-02:
Yeah looks like it's not matching any aliases - needs to match an alias for it to send through to telegram. If you want to send everything to Telegram you could create a wildcard "catch-all" alias and enable Telegram on it :)

closed 2019-01-02 by fletch19