pagermon/pagermon #175: PDW to Pagemon

Shaggs, 2019-01-05:
I have created a python script to allow users to push data from PDW to pagermon.

Is that something @davidmckenzie you would like on here or not?

marshyonline, 2019-01-05:
We have this as a Power-shell script already - a Python script added in there as another option would be good :)
https://github.com/davidmckenzie/pagermon/blob/master/client/psclient.ps1

davidmckenzie, 2019-01-05:
For sure, would be good! Thanks :)

s3m1s0n1c, 2019-01-05:
LOL people still use PDW?  Why when there is so many better programs out there and easier and better at decoding..

112bzo, 2019-01-05:
> LOL people still use PDW? Why when there is so many better programs out there which are easier and better at decoding..

Tell me more please. I use PDW but like to know what kind of software you meen.

s3m1s0n1c, 2019-01-05:
> > LOL people still use PDW? Why when there is so many better programs out there which are easier and better at decoding..
> 
> Tell me more please. I use PDW but like to know what kind of software you meen.

Well Pagermon has its own decoding client in it that does a better job then PDW does..

There are other software but I'm not gonna name them as it takes the light away from this Pagermon which i think is the best out there so far..

Shaggs, 2019-01-05:
> Well Pagermon has its own decoding client in it that does a better job then PDW does..
Not exactly it just uses multimon and rtl_fm using an SH script and uses js to upload.

Multimon is harder to get working on windows and you you are like me and have a desktop scanner that has a discriminator tap then PDW is the best idea.

I have searched for others and multimon and PDW are the only 2 that process FLEX so not 100% correct.

@davidmckenzie ill have it up later today

s3m1s0n1c, 2019-01-05:
I guess we'll agree to disagree then.. 

You could run a Virtual Machine on windows and run Ubuntu and use pagermon that way...
Or run Pagermon on a Raspberry Pi

Each to there own I guess.. It just my opinion..

Shaggs, 2019-01-06:
@s3m1s0n1c seems that way. to me seems a bit silly to run a VM when I could do PDW anywhoo

pull request PDW 2 pagermon #183 contains python file

davidmckenzie, 2019-01-13:
Resolved in #183 😉

closed 2019-01-13 by davidmckenzie