pagermon/pagermon #178: Not all items can be selected

shutterfire, 2019-01-05:
For some reason I can only use a very limited set of icons. Icons with a dash in their name seem to not work at all (heart-rate, briefcase-medical or something like that)

Here's what I tried:
500px works
amilia seems not to work
android works

So the dash thing does not seem to be the issue though

112bzo, 2019-01-05:
I have The same isseu

davidmckenzie, 2019-01-05:
We're on an older version of FontAwesome (4.x), current icons can be seen here: https://fontawesome.com/v4.7.0/icons/

There were a few breaking changes in v5 so may not be simple to upgrade - could break existing aliases. Will need to test upgrading and see :)

shutterfire, 2019-01-05:
Ah, okay. Proposal: Change the "all icons" link to make this clear

closed 2019-01-05 by shutterfire

marshyonline, 2019-01-06:
I've attempted the upgrade in the last week or two with the intent to has it go into master - its not a simple process :( 

I may re visit it in the future tho as i do like the look of the newer icons

DanrwAU, 2019-01-06:
@marshyonline Agreed, i tried it a few months back and gave up as well lol

davidmckenzie, 2019-01-06:
The changes for those curious: https://fontawesome.com/how-to-use/on-the-web/setup/upgrading-from-version-4

We might be able to get away with just using the v4 shims, but it reeks of "this could break without warning at some point" :(