pagermon/pagermon #180: Double quote character is stripped from messages

pyrog, 2019-01-05:
Double quote character `"` is stripped if contained in the message.
This happened when message is stored in database.

Note: If this character is added manually in the database, it is displayed correctly.

davidmckenzie, 2019-01-05:
This is on purpose to reduce risk of SQLi attack, see line 499 in api.js. We could probably change this to replace with a similar char like `”` or something.

davidmckenzie, 2019-01-13:
This may be resolved (or resolvable) when #174 goes to master - the duplicate checking SQL won't be as nasty any more :)

DanrwAU, 2019-01-13:
> This may be resolved (or resolvable) when #174 goes to master - the duplicate checking SQL won't be as nasty any more :)

From reading https://stackoverflow.com/questions/49665023/does-knex-js-prevent-sql-injection Knex does prevent against injection. 
I'll have a play with it and see if i can cause one once i get it all working 👍

DanrwAU, 2019-05-01:
Fixed in #274

PR #274 (2019-05-02): Remove double quote stripping 
# Description

Removes the section of API.js that strips double quotes from messages - Knex protects against SQLInjection so this is no longer required

Fixes #180 

## Type of change

Please delete options that are not relevant.

- [x] New feature (non-breaking change which adds functionality)
- [ ] This change requires a documentation update

# How Has This Been Tested?

Submitted messages containing SQL Statements and confirmed these were NOT executed

# Checklist:

- [x] My code follows the style guidelines of this project
- [x] I have performed a self-review [... cut]
closed 2019-05-02 by DanrwAU