pagermon/pagermon #185: Pushover plugin fails

shutterfire, 2019-01-06:
`uncaughtException: Cannot read property 'value' of undefined
TypeError: Cannot read property 'value' of undefined
    at Object.run (/root/pagermon/server/plugins/Pushover.js:20:34)
    at /root/pagermon/server/plugins/pluginHandler.js:33:30
    at Object.async.forEachOf.async.eachOf (/root/pagermon/server/node_modules/async/lib/async.js:233:13)
    at Object.handle (/root/pagermon/server/plugins/pluginHandler.js:22:11)
    at Statement.<anonymous> (/root/pagermon/server/routes/api.js:582:43)
--> in Database#get('SELECT messages.*, capcodes.alias, capcodes.agency, capcodes.icon, capcodes.color, capcodes.ignore, capcodes.id AS aliasMatch, capcodes.pluginconf FROM messages LEFT JOIN capcodes ON capcodes.id = messages.alias_id  WHERE messages.id = 101', [Function])
    at Statement.<anonymous> (/root/pagermon/server/routes/api.js:569:26)`

Tried with different messages, alias is set perfectly, API-key and userkey are fine and priority is set to normal

DanrwAU, 2019-01-06:
Just on mobile at the moment but it looks like it's complaining about the sound.
Can you try setting a notification sound and seeing if it works?

marshyonline, 2019-01-06:
Unless the API has changed, when you pass it no sound, it uses the default
Sending it a blank sound option my upset it a touch tho. 
I'll take a look today

davidmckenzie, 2019-01-06:
Looks like it's more the plugin that's falling over without the option set, rather than the API, since it's looking for a property that doesn't exist - will put through a quick patch

PR #186 (2019-01-06): do the things to make bad error go away
Fixes #185
/cc @shutterfire
closed 2019-01-06 by davidmckenzie