pagermon/pagermon #187: client on startup

AllanRichardson, 2019-01-06:


AllanRichardson, 2019-01-06:
is there any way to have the client start automagicly after boot?

davidmckenzie, 2019-01-06:
Depends on your system ;) On linux you could add it to rc.local or something.

marshyonline, 2019-01-06:
I do some crazy stuff with it as a system service, its not that hard to get setup :)

DanrwAU, 2019-01-06:
I run it with PM2 on my Pi's it handles auto restarting ;-)

AllanRichardson, 2019-01-06:
ok guys, thanks for your quick responses. im not very experienced with this stuff, but got pm2 to launch the server on start up. currently have 2 clients running. one of the clients is remote and one is local to the server. both running on raspberry pi

AllanRichardson, 2019-01-07:
Got it!!! learned to use rc.local... thanks guys now to figure out how to mark issue closed

closed 2019-01-07 by AllanRichardson

AllanRichardson, 2019-01-08:
pm2 works better