pagermon/pagermon #189: Allow aliasses to contain function letters

eopo, 2019-01-10:
Some regions use function letters not to transmit information about what is going on, but for other usecases. It would be great to be able to store information about function in capcodes in the format 1234567a

davidmckenzie, 2019-01-10:
Not sure what you mean there - could you provide some examples?

eopo, 2019-01-10:
e.G. Adress 1234567 Function 0/a -> Alias 1
Adress 1234567 Function 1/b -> Alias 2
Adress 1234567 Function * -> Alias 3 (Everything that isn't defined more precisely, in this case 2/c and 3/d)

multimon_ng does output this in it's strings with "Function: 0"

davidmckenzie, 2019-01-10:
Ah ok, got ya. I suppose we could merge them together on the client side, e.g. '12345670a' and '12345671b' as the address. The server side will accept alphanumeric addresses on the aliases already, so it's only really changing what data the client side sends over.

Could add a config option to enable sending function code with the address, as I imagine most users wouldn't want it on by default

eopo, 2019-01-11:
Yeah, I'd send it by default though. There are no draws if you have it.
More important is what the aliassing server-side does for me

davidmckenzie, 2019-01-13:
Submitted PR for this in #191 - is this what you were looking for?

eopo, 2019-01-13:
BG information: Usual usecase in Germany would be e.g.
0/a - Fire
1/b - Crash
2/c - Stop
3/d - Test

Other usecases are e.g.
1/a - Everyone
2/b - Use car 1
3/c - Use car 2
4/d - Only chief

A lot of pagers support displaying a predifined text according to the function bits when no text is transmitted in Alpha.

davidmckenzie, 2019-01-13:
Merged in now :)

PR #191 (2019-01-13): add func code to address
Resolves #189

Adds an optional flag to include the function code with the address when sending message to server
closed 2019-01-13 by davidmckenzie