pagermon/pagermon #190: Transferring repo to new organisation

davidmckenzie, 2019-01-13:
When I first created this repo under my username I created another user named 'pagermon' with the view that it would one day take my place as the owner of the repo. This turned out to be a pain in the ass so I ignored it for a while.

Today, [I've converted that account into a GitHub organisation](https://github.com/pagermon) (one of the free ones for open source projects), and will soon be moving the project into it. Nothing should change for contributors or end users, apart from a few new features like the team discussions board. This will also make it easier to knock out #89 

Now that we've got a few more contributors than we used to, I'll also be looking at ways/tools to make collaboration a bit easier, like the aforementioned discussions, reviving the old project thingo,  or maybe starting a Google Groups mailing list.

For now, nothing changes as GitHub [handles transfers pretty neatly](https://help.github.com/articles/transferring-a-repository/), but figured it's worth dropping an issue in here for anyone watching the project.

marshyonline, 2019-01-13:
I wondered why my email filters stoped working 😂
Thanks for the issue letting us know :)

s3m1s0n1c, 2019-01-14:
Trello is pretty good for Project's, You can make sub or subs and chat about each one..
Restrict people to certain projects etc.

Asana is also good and similar

davidmckenzie, 2019-01-14:
The GitHub projects view aint too bad, does the job for now :) https://github.com/pagermon/pagermon/projects/2

closed 2019-01-28 by davidmckenzie