pagermon/pagermon #194: Alias refresh isn't prompted when new alias added

davidmckenzie, 2019-01-13:
Only shows up when updating an existing alias, should appear on both

DanrwAU, 2019-01-13:
I've also noticed this happening when updating, hitting save a second time makes it appear.

It's also not showing on the alias list when required either. 
I had a quick look last night and couldn't figure it out, meant to open an issue...

DanrwAU, 2019-01-18:
This seems to be an issue with the ng-if on line 13 and 18 on aliases.html

Looking at the console, the api is returning the correct value
[image]

However the if is failing to match
[image]

I've tried updating the line to 
`              <button ng-if="settings.database.aliasRefreshRequired == 1" type="button" ng-click="aliasRefresh();" class="btn btn-warning">Refresh Alias/Message Map</button>`

but that didn't work. I know nothing about how this part works so at a bit of a loss with this one 👍