pagermon/pagermon #202: Allow RegEx-Flags

eopo, 2019-01-13:
Allow RegEx-Flags to be set, especially case sensitivity (Tried and failed with that)

davidmckenzie, 2019-01-13:
I mean where was this tried :) The filters plugin should just take it completely raw, it doesn't process the regex at all, so if that's not accepting more complex regex then it's definitely a bug

eopo, 2019-01-14:
I tried it with highlighting

davidmckenzie, 2019-01-14:
Ah I see - yeah the highlighting is kinda pseudo-regex. The way it works is a little dodgy, but is case insensitive by default. E.g. a highlight of `al.rt` will match `ALERT`. Part of the way it works is to clump everything together - it's not running multiple matches.

```js
var regex = new RegExp(tokenized.join('|'), 'gmi');
```

Not really any neat way around that without completely rewriting the library we poached for this purpose.

112bzo, 2019-02-17:
What can i do with this RegEx function?