pagermon/pagermon #206: NotificationsEnabled State not saving

DanrwAU, 2019-01-14:
**Describe the bug**
Cookies are not saving notification state correctly

**To Reproduce**
Steps to reproduce the behavior:
1. Disable browser notifications
2. Close browser and reopen
3. Check website - desktop notifications have been re-enabled

**Expected behavior**
Disabled notifications persist across browser sessions. 

**Screenshots**
Notifications Disabled
[image]

After browser restart
[image]

**Environment information:**
- Browsers Affected: Chrome, Firefox latest
- Pagermon Version: Latest Release + master

**Additional context**
Believe this is due to the expiry being "session" on the cookie. 
Chrome shows expiry date for the field as 
[image]

davidmckenzie, 2019-01-14:
Tested on my prod (few commits back) and dev (latest master) and both setting the cookie to 1969. Updated prod and still setting the expiry correctly.. weird.

Maybe we need to manually define expiry in the $cookies.put function: https://docs.angularjs.org/api/ngCookies/service/$cookies

DanrwAU, 2019-01-19:
Closed from https://github.com/pagermon/pagermon/pull/213 because i have nothing better to do on a friday night 👍

PR #220 (2019-02-04): Release V0.2.3
Release  V0.2.3

* Fix cookies expiring on browser close #206
* Fix refresh aliases button not appearing #194
* Remove solid white background from Icon images #216
* Fix capcode routes not allowing API authentication #218
closed 2019-01-19 by DanrwAU