pagermon/pagermon #212: Make use of control characters

eopo, 2019-01-16:
I've noticed control characters are not implemented but displayed as plain text.
I'd propose to display them as they are intended, especially CR and LF.
EOT could be implemented in a way that any possible text after a EOT would be displayed in a light grey to distinguish it from regular text

DanrwAU, 2019-01-16:
Are you wanting this displayed server side? 

I can't see any example looking through my logs where this would be useful, do you have any examples?

davidmckenzie, 2019-01-16:
From memory they're always bound by <> so could just do a replace on <CR> and <LF> for \n.

We blank out most control codes in the reader:

`message.replace(/<[A-Za-z]{3}>/g,'')`

Don't think I've seen any other than EOT in the wild though.

Re adding things that come after EOT... doesn't that defeat the purpose of EOT? That should denote the end of the message, no?

eopo, 2019-01-17:
I received multiple messages containing CR LF today. Would be cool to see them that way.
Nothing critical though

davidmckenzie, 2019-01-19:
Can you post an example of what the CR/LF codes look like in PDW's raw output?

eopo, 2019-01-27:
I don't know what you meant with PDW, but I finally found one in the log stream (Output from multimon_ng to pagermon)

` ****: ******. 55 a<CR><LF>*****<CR><LF>WB ***** 2. OG  `
(Asteriks are censored content)

DanrwAU, 2019-01-27:
Ah right i should know that. 

Should be a simple replace in reader.js - if you replace <CR><LF> with <br> it works correctly.

davidmckenzie, 2019-01-27:
It'd be nice to replace it with `\n` and `<pre>` the message, but that would break highlighting. Replacing with `<br>` at the reader end seems to be best solution without dodgy hackery.

LRacing, 2020-04-14:
> Ah right i should know that.
> 
> Should be a simple replace in reader.js - if you replace `<CR><LF>` with `<br>` it works correctly.
> 
> [image]

Hello from Sweden and thanks for a well functioning program.
I have tested as above and it looks good in sqlite and when it first pops up in the browser with line breaks and no incorrect characters, but if you update the page then the line break disappears and it says br without <> so filtering seems to be done when the page is updated. thoughts on this?
[image]
[image]

wtc-brycel, 2022-06-11:
> Hello from Sweden and thanks for a well functioning program. I have tested as above and it looks good in sqlite and when it first pops up in the browser with line breaks and no incorrect characters, [...]

I am still seeing the same behavior as LRacing. Unfortunately I am just replacing <LF> in reader.js with "---" to separate portions of the message, but it looks pretty bad. Our system makes heavy use of the <LF> control character in our fire pages.