pagermon/pagermon #216: Update favicon have transparent background

davidmckenzie, 2019-01-28:
**Describe the solution you'd like**
Switched to firefox recently, dark mode makes the white background of the favicon super obvious. The white bit outside of the bullhorn should be transparent.

**Describe alternatives you've considered**
Alternative is to ignore it because it's a favicon, who cares

**Additional context**
Probably see if I can knock out #50 at the same time

DanrwAU, 2019-01-29:
I can handle this one this afternoon if need be - is this what your after?
[image]

davidmckenzie, 2019-01-29:
Yah was thinking that but with the inside of the bullhorn filled, so it's just the outside white that's gone. Thanks dude <3

DanrwAU, 2019-01-29:
No worries, i'll knock them all out when i get home 😉

closed 2019-01-29 by DanrwAU