pagermon/pagermon #223: Error 404 sending to remote server

dh47, 2019-02-06:
I have setup my pi as the client and I have a remote connection to the server side, all is well but I'm having issues sending the data to the remote server.

https://pastebin.com/skYsq0wk

Thanks.

davidmckenzie, 2019-02-06:
What's your `hostname` set to in your client's config file?

marshyonline, 2019-02-06:
Looks to me that it's hitting the home page and not the API 
You need to make sure that HOSTNAME is the client config is set to 
```http://your-ip-or-domain/api/messages

```

dh47, 2019-02-06:
"hostname": "http://XX.63.26.185:3000/",

DanrwAU, 2019-02-06:
As per Marshy's comment (which i just fixed lol) you need to put the full path to the api in there. 

so it would be `"hostname": "http://XX.63.26.185:3000/api/messages"`

dh47, 2019-02-06:
Wow, things that can happen when you add a "/" at the end.

davidmckenzie, 2019-02-06:
Yeaaah that shouldn't happen... it should be smart enough to deal with that :D Will reopen and fix!

DanrwAU, 2019-02-07:
Should be able to do soemthing like this

```
function stripTrailingSlash(hostname) {
    if(hostname.substr(-1) === '/') {
        return hostname.substr(0, hostname.length - 1);
    }
    return hostname;
}
```

dh47, 2019-02-20:
Just wanted an update on this?  Has this been pushed?

DanrwAU, 2019-02-20:
I haven't seen a commit for it yet. 

Unfortunately @davidmckenzie @marshyonline and myself have all been busy with life stuff recently. Personally i haven't really had a chance to do much of anything with pagermon recently.

marshyonline, 2019-02-20:
I don't think we have pushed a fix to strip the tailing slash
Quick way to fix on your end is to remove the trailing slash from the hostname

dh47, 2019-02-20:
Okay.

DanrwAU, 2019-03-16:
This is fixed in https://github.com/pagermon/pagermon/pull/233

DanrwAU, 2019-04-26:
This will be fixed in v0.3.0 ... closing 😁

PR #233 (2019-03-16): Fix 223 - Add validation for trailing slashes
# Description

Add's validation to ensure no trailing slash on the URI in the config file

Fixes #223 

## Type of change

Please delete options that are not relevant.

- [x] Bug fix (non-breaking change which fixes an issue)

# How Has This Been Tested?

- [x] Tested Message submission from reader.js after implementing
- [x] Tested to ensure URI is correct both with and without a trailing slash in the config file

# Checklist:

- [x] My code follows the style guidelines of this project
- [x] I have performed a self-review of my own code
- [x] I ha [... cut]
closed 2019-04-26 by DanrwAU