pagermon/pagermon #224: Upgrade to 0.2.3

112bzo, 2019-02-10:
Hello, I have now 0.2.0 working. Is it possibel to overwrite de files zo i can use 0.2.3 or must i do all again new.

So question: can i override the files?

marshyonline, 2019-02-10:
The update process is fairly simple.
Take a backup of your message.db and config.json and you can essentially overwrite the rest of the files

davidmckenzie, 2019-02-10:
To add to this - it *should* be entirely safe to just completely overwrite the current directory with the new release. The config and database files aren't included in the zip so shouldn't be overridden, but always take backups as a precaution :)

DanrwAU, 2019-03-16:
This one's been open a month, if your having any dramas come see us in discord and we can help you out 👍

closed 2019-03-16 by DanrwAU