pagermon/pagermon #225: Add info message to top of page

davidmckenzie, 2019-02-16:
**Describe the solution you'd like**
Allow admin to add a custom info banner at the top of the page to communicate maintenance, changes, etc.

Consider adding functionality to have a 'more info' button that opens modal.

Also consider adding expiry for the message.

s3m1s0n1c, 2019-02-16:
This would be cool function to have.

At the moment the only way is to use testMessage.sh and inject custom messages into the DB and have my own custom Capcode in Alias.
👍

DanrwAU, 2019-03-16:
Maybe a status page - show server and client stats somehow as well?

DanrwAU, 2020-06-03:
@pagermon/collaborators are we happy with the one-time modal for this?

marshyonline, 2020-06-03:
I think one-modal fits this well

davidmckenzie, 2020-06-03:
Yah, theme can fill role of anything else