pagermon/pagermon #229: Switching to page 1 doesn't update URL

davidmckenzie, 2019-02-27:
**Describe the bug**
Switching to page 1 doesn't update URL

**To Reproduce**
Steps to reproduce the behavior:
1. Go to main page
2. Click on page 2
3. Observe URL updates
4. Click on page 1
5. Observe URL stays at page 2

**Expected behavior**
URL should update to page=1

DanrwAU, 2019-03-16:
This might be a mac thing, this works correctly for me in chrome, both using the back button in the browser and using the pagination buttons resets the url bar to the root url when going back to page 1

davidmckenzie, 2019-03-17:
Works on Chrome, but not Safari or Firefox it seems. It's likely a thing with those browsers not liking "nothing" as the parameters - will update it to change to page=1 when you navigate back to page 1.

PR #236 (2019-03-18): Url update
# Description

Update the pushState bits to properly work with firefox - it doesn't like blank things. Also stops duplicate states from being inserted, not sure if this was an issue in Chrome but was creating many history states in Firefox

Fixes #229 

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)

# How Has This Been Tested?

Browse on many browsers, switch pages, observe changes

# Checklist:

- [x] My code follows the style guidelines of this project
- [x] I have performed a self-review of my own code
- [x] I have commented my code, particularly in  [... cut]
closed 2019-03-18 by davidmckenzie