pagermon/pagermon #230: Alias Groups

eopo, 2019-03-07:
Hi, I'd like to group aliasses.
This could be done either manually (Create group, select aliasses, add) or automatically (Create group, add all aliasses containing information whatever (E.g. address matching 123%, Description starting with "XYZ", Organization matching "Fire", ...) or even both (❤️)

Groups could be able to override alias settings and push to plugins.

davidmckenzie, 2019-03-18:
This would be neat, bundle it in with #204 or something.

Would be useful for me too - I'd like to group my aliases by district as well as agency.

eopo, 2019-03-25:
Oh: And filtering by groups would be cool as well

eopo, 2019-07-09:
How about Auto create groups for same alias

s3m1s0n1c, 2020-07-07:
This would be cool to do as a "User Settings" option :)  and would benefit my users as well..