pagermon/pagermon #238: On and off function

112bzo, 2019-03-23:
I would like to see that I could also switch off the Agency and the filter with a selection button

so only in the front end i see the date, time, icon and message. NOT the agentcy and filter tekst.

How to?

DanrwAU, 2019-03-24:
Agency is a required column in the database so it would have to be there on the backend. 
If you just don't want to SEE it, you could edit index.ejs to remove the following from line 70-77.

```
<td class="shrink clickable-td" ng-if="message.agency" uib-popover-template="'templates/popover.html'" popover-append-to-body=true ng-click="popoverEl = 'agency'" popover-trigger="'outsideClick'">
              <span class="fa-stack fa-lg fa-fw" style="color: {{message.color || 'grey'}};">
                <i class="fa fa-circle fa-stack-2x"></i>
                <i ng-if="message.icon" class="fa fa-{{message.icon}} fa-stack-1x fa-inverse"></i>
                <i ng-if="!message.icon" class="fa fa-question fa-stack-1x fa-inverse"></i>
              </span>
              <strong style="vertical-align: middle;">{{message.agency || ''}}</strong>
            </td>
```

To remove the filter popover from the address you would need to edit line 86, which would depend on if you wanted to popover removed entirely, or just for non-logged in users.

112bzo, 2019-03-24:
Seems good, thanks mutch better.
[image]

Great job

DanrwAU, 2019-03-24:
Glad it solved your issue, i was merely tinkering for my own benefit 🤣 🤣

closed 2019-03-24 by DanrwAU