pagermon/pagermon #248: Accessing api from Node-red

ggreer123, 2019-03-28:
I'm setting up a node-red and want to use the output of node-red to turn a light on (tragic) when a particular capcode is sent. I've looked, but can't figure out how to access the api. Can anyone help?

Thanks

marshyonline, 2019-03-28:
I have no experience with node red but you would basicly need to write a pagermon plug into forward the messages to an API running on your end 

There is a PR open for a plugin that triggers shell commands - this might be of use for you

davidmckenzie, 2019-03-28:
Node-red looks pretty neat. From a quick google, I wonder if you could use the Websockets things: http://noderedguide.com/node-red-lecture-3-example-3-7-using-websockets-with-node-red/

Configure a websocket input based on the wss url used in Pagermon (you could find this in your browser's dev tools)

Like Marshy I got 0 experience with node-red so this is just from a bit of quick googling :)

ggreer123, 2019-03-28:
Thanks, I'll have a look. I use node red with home assistant, and if I get it going, I'll share the flow.Gary

DanrwAU, 2019-03-28:
No experience here either. 

I think i've still got a Pi floating around at home not doing anything, i might see if i can come up with a plugin over the weekend 😁. Need to read the API docs first and see whats possible tho so no promises 😉

closed 2019-05-26 by marshyonline