pagermon/pagermon #250: Winston is in love with minute 4

DanrwAU, 2019-04-03:
**Describe the bug**
Winston seems to have an obsession with the 4th minute of every hour. 

When the hour ticks over, winston will report the time as XX:04 where xx is the hour. 

**To Reproduce**
Steps to reproduce the behavior:
Read your logs, see logs like this. 
1|pagermon | 2019-04-03 17:39 +11:00: [pmon]  2019-04-03 17:04:86 

**Expected behavior**
Actual real times 

**Environment information:**
 - OS: Encountered on Windows and Linux
- Pagermon: Seen in v0.2.3 and above.

marshyonline, 2019-04-03:
Replicated in two of my prod env's :sadpepe:

DanrwAU, 2019-04-03:
@davidmckenzie any idea's with this one we've confirmed it's occuring in both v0.2.3 and v0.3.0.

DanrwAU, 2019-04-03:
Fixed in #251

closed 2019-04-03 by DanrwAU