pagermon/pagermon #256: SQLite - Duplicate Message during alias refresh

DanrwAU, 2019-04-12:
**Describe the bug**
Duplicate Message checking does not seem to work if messages are submitted while AliasRefresh is running. 

**To Reproduce**
Steps to reproduce the behavior:
Start long running aliasRefresh cycle and submit duplicate messages

**Expected behavior**
Duplicate messages filtered correctly. 

**Additional context**
Tested only in v0.2.3

davidmckenzie, 2019-04-12:
I'm surprised messages even submit at all during the refresh tbh

DanrwAU, 2019-04-12:
Me too, they seem to hold and be processed later ?

DanrwAU, 2019-05-01:
I guess this could be related to #264 in that they are all submitted at the same time after the aliasRefresh finishes?

DanrwAU, 2021-01-22:
I haven't seen this one since. closing as fixed i guess?

closed 2021-01-22 by DanrwAU