pagermon/pagermon #264: v0.3.0 - Duplicate Checking not working

DanrwAU, 2019-04-26:
**Describe the bug**
Duplicate checking not working when message limit is set to 0

**To Reproduce**
Set duplicate limit to 0, duplicate time to 60. 
Submit multiple copies of the same message (more than 2) 
Only the first is filtered, subsequent messages are submitted

**Expected behavior**
All messages are filtered correctly

**Additional context**
This **MAY** have already been fixed with the commits i've made but intermittently still seeing the issue. 
Issue for visibility. 
This one should block release of 0.3.0 for now.

DanrwAU, 2019-04-27:
I'm thinking this may be an issue with incorrect time's on some of my receivers. 

I'll leave this open incase anyone else experiences this with v0.3.0

DanrwAU, 2019-04-27:
Not sure what to do with this one, Problem seems to be messages that are being submitted at exactly the same time are not matching.

This wasn't an issue with v0.2.3 for some reason. 

Setting the message limit seems to make the problem worse.

@davidmckenzie  mind casting an eye over lines 530-573 in api.js ?

Also @pagermon/collaborators should we block v0.3.0 for this one?

DanrwAU, 2019-04-27:
This seems to be related to the Timestamp calculations when the times are equal or close. 
Changing to only limit mode seems to solve the issue
[image]

davidmckenzie, 2019-04-27:
When you say exactly the same time... how exactly are we talking? Is it two separate receivers sending simultaneously or the sequential sending of a single receiver that gets two dupes in a row?

Guessing this is on MySQL too? Could be race condition just from MySQL being multithreaded. The insert and alias matching would take a few ms that might be enough time for the second message to start its insert too.

Could add a second layer of dupe checking - an array that holds the last x messages, but that wouldn't help if we go full multithread/clustered/load balanced. Good enough workaround for future problem solving with a proper MQ

DanrwAU, 2019-04-27:
Multiple receivers, SQLite. 
Exactly the same second.... 

In testing it does see to pick up they are duplicates when using message count only mode, i think the problem is with this part of the querries

.where('timestamp', '>', timeDiff)

But i'm not sure whats wrong with it and i havent had enough sleep to begin figuring it out 😢

davidmckenzie, 2019-04-27:
Hmm what about when it's only time limit and no message limit?

DanrwAU, 2019-04-27:
Worse ... it never matches.
With both it's sometimes and with only message limit it works perfectly.

*head explodes*

DanrwAU, 2019-04-27:
Never mind just had one duplicate with limit only.

I quit...

davidmckenzie, 2019-04-27:
Well that's weird... are both the clients running latest copy? Maybe something weird with the timestamp parsing on client side. No hard to find whitespace on the dupe that was on limit only? Will have to do some sciencing I think

DanrwAU, 2019-04-27:
All receivers are new reinstalls running identical bastardized versions of reader.sh for multifm.
:(

DanrwAU, 2019-04-28:
Looked further into this today, i'm really at a loss, all the queries and bindings look correct when using the knex debugging. 
Only thing i can think of is that the messages are being inserted at exactly the same time (or close enough too).  
I'm wondering if i can build a delay into reader.js on each receiver to stagger the submissons. 
This shouldn't be an issue for duplicate messages being sent to pagermon by the same receiver, just my specific wierd use case.

DanrwAU, 2019-04-28:
Confirmed that adding a 100ms delay between each receiver resolves the issue. 
(one firing at message receival, one at 200ms, one 300ms, one 400ms and one 500ms).

Maybe something to add to a known issue list and move on with release? @pagermon/collaborators opinions?

marshyonline, 2019-04-28:
I would be against a random <=1s delay for messages

DanrwAU, 2019-04-28:
Not sure it's required outside of this very specific use case.  
Possibly a setting in the reader config?

davidmckenzie, 2019-04-28:
This might still help

> Could add a second layer of dupe checking - an array that holds the last x messages

Adds fair bit of complexity and doesn't solve the problem completely but would bring the race condition potential much lower.

Client side might be better - could add a 'rank' flag in the config. Rank 1 client is the primary feeder and goes instantly, 2 has 100ms delay, 3 has 200ms, etc.

Have you tried playing with how low you can drop the delay?

I'm also curious why this only started happening now. The only thing between the dupe check and the insert is the capcode identification bit (579-583). Maybe the knex version of that query is taking slightly longer to process.

DanrwAU, 2019-04-28:
Or the non-knex is waiting for the previous submission to finish before starting?

Not sure

davidmckenzie, 2019-04-28:
Good point, that sounds more likely

davidmckenzie, 2019-04-28:
Could it need to be wrapped in a transaction? https://knexjs.org/#Transactions

DanrwAU, 2019-04-28:
To be fair they possible should have been used to replace all the db.serialize commands ....

Oops?

DanrwAU, 2019-07-11:
I've confirmed that messages including a \ always fail the duplicate check.

DanrwAU, 2019-08-14:
Resolved in #303

PR #301 (2019-08-13): Optimize Duplicate checking queries
# Description

Possible fix for the duplicate message checking issues introduce in v0.3.0 and Knex when using multiple receivers. 

Doesn't Fix #264 

## Type of change

Please delete options that are not relevant.

- [x] Bug fix (non-breaking change which fixes an issue)

# How Has This Been Tested?

Running this in one of my production environments and seeing significantly better dupe checking. Would like feedback from others who were seeing this issue before calling it fixed though.

@s3m1s0n1c can you try this one out and see if your still seeing dupes?
 [... cut]
PR #302 (2019-08-14): Dupey McDupe
# Description

Why waste time say lot word when few word do trick?

Fixes #264

## Type of change

Please delete options that are not relevant.

- [x] Bug fix (non-breaking change which fixes an issue)

# How Has This Been Tested?

Tested a few different dupe settings, needs testing from those suffering from the issue

# Checklist:

- [x] My code follows the style guidelines of this project
- [x] I have performed a self-review of my own code
- [x] I have commented my code, particularly in hard-to-understand areas
- [x] I have updated changelog.md with changes made
closed 2019-08-14 by DanrwAU