pagermon/pagermon #265: v0.3.0 - Sqlite3 - Cannot read property 'prototype' of undefined

s3m1s0n1c, 2019-04-27:
**Describe the bug**

when running node app.js I get the following error.

``$ npm install sqlite3 --save
Cannot read property 'prototype' of undefined
TypeError: Cannot read property 'prototype' of undefined
    at inherits (/opt/pagermon/server/node_modules/sqlite3/lib/sqlite3.js:27:16)
    at Object.<anonymous> (/opt/pagermon/server/node_modules/sqlite3/lib/sqlite3.js:66:1)
    at Module._compile (module.js:643:30)
    at Object.Module._extensions..js (module.js:654:10)
    at Module.load (module.js:556:32)
    at tryModuleLoad (module.js:499:12)
    at Function.Module._load (module.js:491:3)
    at Function._load (/opt/pagermon/server/node_modules/pmx/lib/transaction.js:94:21)
    at Module.require (module.js:587:17)
    at require (internal/module.js:11:18)
    at Client_SQLite3._driver (/opt/pagermon/server/node_modules/knex/lib/dialects/sqlite3/index.js:44:12)
    at Client_SQLite3.initializeDriver (/opt/pagermon/server/node_modules/knex/lib/client.js:253:26)
    at Client_SQLite3.Client (/opt/pagermon/server/node_modules/knex/lib/client.js:84:10)
    at new Client_SQLite3 (/opt/pagermon/server/node_modules/knex/lib/dialects/sqlite3/index.js:31:19)
    at Knex (/opt/pagermon/server/node_modules/knex/lib/knex.js:57:42)
    at Object.<anonymous> (/opt/pagermon/server/knex/knex.js:2:33)
Error: Knex: run``

Its related to **server/node_modules/sqlite3/lib/sqlite3.js - Line 27**

``function inherits(target, source) {
    for (var k in source.prototype)
        target.prototype[k] = source.prototype[k];
}``

Target is returning NULL

So i added below check for NULL

``function inherits(target, source) {
    if (target == null)
        return;
 
     for (var k in source.prototype)
        target.prototype[k] = source.prototype[k];
}``

Not sure why its returning NULL tho..

I'm no node expert :)

davidmckenzie, 2019-04-27:
That looks like bug in sqlite3 npm module? May need to raise over there

s3m1s0n1c, 2019-04-27:
Lol yep

Just looked thru there issues
https://github.com/mapbox/node-sqlite3/issues/1130

Will close this off :)

closed 2019-04-27 by s3m1s0n1c