pagermon/pagermon #267: Make capcode refresh great again

davidmckenzie, 2019-04-27:
If we know the aliases that have changed since last refresh, then we can drastically reduce the number of messages that we need to scan.

Move the refresh required flag to a column on the alias table.

Capcode refresh then looks for all messages that:

a) have that alias id associated, or
b) have a higher level alias associated (e.g. I'm adding an alias for 123123 and I already have an alias for 12%, so include those associated messages), or
c) have no alias attached

C would probably be a separate query, this also might help speed things up by splitting up the logic.

Now what am I missing...

davidmckenzie, 2019-04-27:
Could also use a 'refresh type' variable - new/updated or something.

The searching for higher level aliases should be simple - sub out the special chars and search on that. That way it fits with any arbitrary length of capcode that might be used in exotic systems.

Inserting a new alias: The only messages that need updating are those that have no existing alias and match the new alias, and those that are associated to a higher level alias.

Deleting an alias: This is even easier, they just need existing association removed and to search for higher level aliases. Pretty much same logic as currently used but only on messages with this alias.

Updating an alias: This is pretty much just doing both the insert and delete ones. You need to find all of the current associated messages and reassociate them with whatever, and you need to look at the new matches like when inserting a new alias.

DanrwAU, 2019-04-30:
This is basically what i was thinking as well, but more well structured and sense making 👍