pagermon/pagermon #276: Question : Receive tone only messages

all3kcis, 2019-05-04:
Hi,
I think my question is not pagermon dependant but I would like to know how to receive "tone only" messages.
I would like to anticipate the reception of alpha text messages. The tone only coming in a few seconds before.

Do you have any information to help me?

Thx

davidmckenzie, 2019-05-05:
I forget how these manifest in multimon but should just be as simple as changing the client script to filter differently. As long as `message`, `address`, and `trimMessage` variables are present, it'll send the message. So you'd just have to match on the right text and inject something as the "message".

all3kcis, 2019-05-06:
Thank you for your answer, that's what I was also thinking, but I don't understand the output of multimon.
I used multimon like  
` multimon-ng -t raw -e -p -c -a POCSAG512 -a POCSAG1200 /dev/stdin | node ./reader.js >> /test.log &` to debug ( removed "-f alpha" flag) but I'm not getting anything conclusive.

I only receive alpha messages or unreadable numeric messages .
```
2019-05-06 02:44:48: 0680195:  <message text well decoded>
2019-05-06 02:44:48: POCSAG1200: Address:       2  Function: 0  Numeric: 28480
```

First, I will update multimon-ng to 1.1.7 and test adding flex decoding.

all3kcis, 2019-05-07:
It's ok, just remove "-e" flag ... (remove empty messages)
Tone Only is empty message... 
Ex : `2019-05-07 16:01:00: 0120158: POCSAG1200: Address:  120158  Function: `

closed 2019-05-07 by all3kcis