pagermon/pagermon #278: Filter Loop when using ? in alias

s3m1s0n1c, 2019-05-07:
When having a ? in the alias name this can cause a huge loop when doing a filter/search query on the alias name

when I do a search query on https://compagers.rfspager.com/?q=Dungog? it search's then flaps between that and https://compagers.rfspager.com/?q=Dungog?%3F

This causes MASSIVE load on the server and will end up killing the server or node app if using PM2 if left long enough

davidmckenzie, 2019-05-07:
Welp, that's bad... tested and confirmed

DanrwAU, 2019-05-07:
Probably a bug in the way I rewrote the search on v3 unless you tested in 2.3?

davidmckenzie, 2019-05-07:
It's probably this bit:

```
        // get the current query string - if it's blank or matches the one we're trying to set to, do nothing. This prevents duplicate history entries
        var curQuery = window.location.search;
        if (curQuery == '')
          curQuery = '/';
        if (curQuery != qString)
          window.history.pushState('', '', qString);
```

s3m1s0n1c, 2019-05-07:
I originally tested in V0.3.0 where the problem exists.

The issue doesn't exist in V0.1.8
https://www.rfspager.com/?q=Dungog? 
gets converted to 
https://www.rfspager.com/?q=Dungog%3F

Something to do with encoding ? into %3f

s3m1s0n1c, 2019-05-07:
m4r5hyToday at 3:03 PM
2.0 beta is a bit unhappy about it
eventully it pops over tho

davidmckenzie, 2019-05-07:
Got a fix coming

davidmckenzie, 2019-05-07:
#279 with the fix - lets push out 0.3.1 once that's merged since it's security vuln

s3m1s0n1c, 2019-05-07:
Confirmed #279 fixes the issue

PR #279 (2019-05-07): encode uri components
# Description

Resolves issue with URI encoded bits of search strings not matching the current URL

Fixes #278

## Type of change

Please delete options that are not relevant.

- [x] Bug fix (non-breaking change which fixes an issue)

# How Has This Been Tested?

Quick tests in browser only, looks better

# Checklist:

- [x] My code follows the style guidelines of this project
- [x] I have performed a self-review of my own code
- [x] I have commented my code, particularly in hard-to-understand areas
- [x] I have updated changelog.md with changes made
closed 2019-05-07 by DanrwAU