pagermon/pagermon #285: 0.3.1 Injecting/Displaying messages with wrong timestamp

kiwiscout, 2019-05-12:
I'm hoping its not something I have done prior... or a local environment variable
Any help to rectify appreciated.
Only since updating last night messages rec since midnight are being displayed as last night..

**Describe the bug**
Data displaying with time 12hr out

**To Reproduce**
Monitoring reader.sh injection, can see messages being passed
Monitoring process.json can see messages being processed
Webpage appears to show them 12hrs prior

**Expected behavior**
More messages than current timestamps (comparing with secondary source also)

**Screenshots**
Example of passed hit not displayed - sanitized output.
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : beforeMessage start
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : ======================
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : trigger: message scope: before
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : ======================
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : data object
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : { address: '001920579',
0|pagermon |   message: 'STRU <snip> (XStr ) ',
0|pagermon |   datetime: '1557573892',
0|pagermon |   source: 'SDR',
0|pagermon |   pluginData: {} }
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : plugins object
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : { Pushover: { enable: false, pushAPIKEY: '' },
0|pagermon |   SMTP:
0|pagermon |    { enable: true,
0|pagermon |      mailFrom: 'gmail.com',
0|pagermon |      mailFromName: 'Pagermon',
0|pagermon |      server: 'smtp.gmail.com',
0|pagermon |      port: '465',
0|pagermon |      username: 'gmail.com',
0|pagermon |      password: '',
0|pagermon |      secure: true } }
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : ======================
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : ======================
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : plugin: Pushover
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : ======================
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : plugin: SMTP
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : Plugin does not run in this scope
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : { address: '001920579',
0|pagermon |   message: 'STRU 501 <snip> (XStr ) .
0|pagermon |   datetime: '1557573892',
0|pagermon |   source: 'SDR',
0|pagermon |   pluginData: {} }
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : beforeMessage done
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : afterMessage start
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : ======================
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : trigger: message scope: after
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : ======================
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : data object
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : { id: 267452,
0|pagermon |   address: '001920579',
0|pagermon |   message: '<snip> (XStr )',
0|pagermon |   source: 'SDR',
0|pagermon |   timestamp: 1557573892,
0|pagermon |   alias_id: 1519,
0|pagermon |   alias: 'Op Support',
0|pagermon |   agency: '',
0|pagermon |   icon: 'fire',
0|pagermon |   color: 'red',
0|pagermon |   ignore: 0,
0|pagermon |   pluginconf:
0|pagermon |    { Discord: { enable: false, webhook: '' },
0|pagermon |      Pushover:
0|pagermon |       { enable: false,
0|pagermon |         priority: { value: '' },
0|pagermon |         group: '',
0|pagermon |         sound: { value: '' } },
0|pagermon |      SMTP: { enable: false, mailto: '' },
0|pagermon |      Telegram: { enable: false, chat: '' },
0|pagermon |      Twitter: { enable: false, hashtag: '' } },
0|pagermon |   aliasMatch: 1519,
0|pagermon |   pluginData: {} }
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : plugins object
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : { Pushover: { enable: false, pushAPIKEY: '' },
0|pagermon |   SMTP:
0|pagermon |    { enable: true,
0|pagermon |      mailFrom: 'gmail.com',
0|pagermon |      mailFromName: 'Pagermon',
0|pagermon |      server: 'smtp.gmail.com',
0|pagermon |      port: '465',
0|pagermon |      username: 'gmail.com',
0|pagermon |      password: '',
0|pagermon |      secure: true } }
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : ======================
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : ======================
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : plugin: Pushover
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : ======================
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : plugin: SMTP
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : RUNNING PLUGIN!
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : { id: 267452,
0|pagermon |   address: '001920579',
0|pagermon |   message: '<snip> (XStr BEA) ',
0|pagermon |   source: 'SDR',
0|pagermon |   timestamp: 1557573892,
0|pagermon |   alias_id: 1519,
0|pagermon |   alias: 'Op Support',
0|pagermon |   agency: '',
0|pagermon |   icon: 'fire',
0|pagermon |   color: 'red',
0|pagermon |   ignore: 0,
0|pagermon |   pluginconf:
0|pagermon |    { Discord: { enable: false, webhook: '' },
0|pagermon |      Pushover:
0|pagermon |       { enable: false,
0|pagermon |         priority: { value: '' },
0|pagermon |         group: '',
0|pagermon |         sound: { value: '' } },
0|pagermon |      SMTP: { enable: false, mailto: '' },
0|pagermon |      Telegram: { enable: false, chat: '' },
0|pagermon |      Twitter: { enable: false, hashtag: '' } },
0|pagermon |   aliasMatch: 1519,
0|pagermon |   pluginData: {} }
0|pagermon | [pmon]  2019-05-12 11:24:52  debug : afterMessage done

2019-05-11 | 23:24 | SDR | 001920579 | Op Support | STRU (XStr BEA) -- | -- | -- | -- | -- | -- | --

**Environment information:**
 - OS: Pi - raspian /debian 9 stretch
 - Other details:

**Additional context**
Perhaps only 1600/FLEX times, as standard pocsag from ambulance service appear to be normal

DanrwAU, 2019-05-12:
Do the messages include a timestamp in the message?

Possibly related to this?
https://github.com/pagermon/pagermon/pull/235

kiwiscout, 2019-05-12:
> Do the messages include a timestamp in the message?
> 
> Possibly related to this?
> #235

Standby, I'll change back to console output and grab the entire string from reader.sh

kiwiscout, 2019-05-12:
Don't appear to have one in them thats visible in any of the outputs to/from reader

But I do note when it's injected, it momentarily shows up at the top of the view update, but once you refresh re-orders it to -12hrs
|pagermon | ::ffff:127.0.0.1 - - [12/May/2019:00:33:24 +0000] "POST /api/messages HTTP/1.1" 200 6 "-" "PagerMon reader.js"

I did note that client/app,js is lacking a re-form? line under the FLEX section that pocsag has
 message = message.replace(/\d{2} \w+ \d{4} \d{2}:\d{2}:\d{2}/,'');

Not related to adjsting the time by chance?

davidmckenzie, 2019-05-12:
Is timezone same on both the client box and the server box?

kiwiscout, 2019-05-12:
> Is timezone same on both the client box and the server box?

Yes sir, Same device/RPi

marshyonline, 2019-05-12:
I updated form v0.2.3 lastnight to v0.3.1 and am not seeing this issue with POCSAG messages(Ignoring the message timestamp)

I would need someone to send me raw Multimon Output or RAW rf recordings to have a play as no flex here :(

kiwiscout, 2019-05-12:
That may be possible. I just modified the FLEX else datetime moment(timestring) matching the check above it and managed to make it stay at the top with INVALID TIME. But that's less than ideal

Quite regular FLEX during daylight hours here. Any way to make it STDOUT while capturing as normal

kiwiscout, 2019-05-12:
Nevermind, Removing node reader.js from the pipe did it

This help? Have to sanitize it a bit

FLEX: 2019-05-12 01:45:28 1600/2/A 11.056 [000999382] ALN (<truck id>) <message>

kiwiscout, 2019-05-12:
I've stripped the added flex address modifier from reader.js from pull for the time being. As it appeared to be ignoring true/false in config,json and still doing it. Now back to dumping entire message with rec time.

Let me know if you need a sample dump of really bad audio (and perhaps how to grab) or more raw flex messages

davidmckenzie, 2019-05-12:
Is that a UTC timestamp in there? Sounds like it's reading that UTC time and interpreting it as local time

kiwiscout, 2019-05-12:
Yep, the flex codes here appear to get UTC stamps if that's a good extract. Example above was after lunch. So when using that and not rec time it seemed to put itself the night before.

Guess that introduces a bunch more code to need to convert UTC to local :/

DanrwAU, 2019-05-12:
You could always run the Pi in UTC timezone. 
The browser displays in local TZ of the browser anyway.

davidmckenzie, 2019-05-12:
I wonder if that's _always_ UTC... any other FLEX users that can confirm? If so then easy enough to make the moment thing get the unix timestamp in utc, if it varies then... too hard basket 😁

kiwiscout, 2019-05-13:
Got any pals in SA? Apparently they too use FLEX for emergency paging?

Theres an audio sample here https://www.sigidwiki.com/wiki/FLEX that I was going to try and decode and see if the head was in UTC or if it is local implementation that adds it.

kiwiscout, 2019-05-13:
} else if (line.match(/FLEX: \d+-\d+-\d+ \d{2}:\d{2}:\d{2}/)) {
        timeString = line.match(/\d+-\d+-\d+ \d{2}:\d{2}:\d{2}/)[0];
        if (moment(timeString).isValid()) {
          datetime = moment(timeString)**.add(12, 'hours')**.unix();
        }

Seems to do the trick too. (at least on the periodic test FLEX messages - real one TBA) So I guess can make a config.json TZ type variable to specify and insert in place of the manual change there. 
Or is there a nicer code build in to node to do UTC-> local

kiwiscout, 2019-05-23:
Well the above adjustment works fine for all FLEX it seems. 
But also found you can specify the input is UTC and have it convert to localtime by adjusting to 
datetime = moment.utc(timeString).unix(); which may take care of DST changes.

The odd thing, I have useTimestamp set as True, but appears to be using the 'false' else if of the flex section.. is that expected or am I reading true/false section backwards

GerritS76, 2019-05-27:
>       datetime = moment(timeString)**.add(12, 'hours')**.unix();

Didn't seem to work for me, however ... this one did work for me.
          datetime = moment(timeString).add(12, 'hours').unix();

kiwiscout, 2019-05-27:
> > ```
> >   datetime = moment(timeString)**.add(12, 'hours')**.unix();
> > ```
> 
> Didn't seem to work for me, however ... this one did work for me.
> ` datetime = moment(timeString).add(12, 'hours').unix();`

Yeah, Sorry, the ** are default insert for Bold to highlight the change, it didn't go as expected.

So not the only one finding they are in UTC then, AU or NZ based? it may tell us if its a global thing or national system implimentation

takenalias, 2019-05-28:
I changed it to datetime = moment.utc(timeString).unix(); and it worked a treat. I'd suggest doing that @GerritS76

kiwiscout, 2019-05-28:
Was beginning to think I was alone for a while. Good to see it wasn't my setup specific

I guess it raises the question do we request a merge with .utc moment translation hardcoded as 'the fix', or make a toggle/timezone from config with more if statements looking out for such a variable to futureproof or count for other FLEX that may run with localtime..

davidmckenzie, 2019-05-28:
I wouldn't assume that it's the same everywhere - will need to be config, should be simple enough :)

kiwiscout, 2019-05-28:
I'll admit, I did have a hack at it. But failed miserably when it come to the small var function to define the adjustment by way of user input. Not enough cookie smarts. I had it as far as looking for a UTC | true/false but then broke the additional if under the timeStamp one.

GerritS76, 2019-05-28:
> I changed it to datetime = moment.utc(timeString).unix(); and it worked a treat. I'd suggest doing that @GerritS76
Thanks for the suggestion, changed it into moment.utc which works like a charm ;-)
Cause it was quite late, I stopped reading about the moment.tz which might have been a solution tho?

ULTRUM, 2020-02-19:
Is anyone still having issues with this? I've been playing around, but it seems that when I change the reader.js to `datetime = moment(timeString).add(12, 'hours').unix();` or even `datetime = moment.utc(timeString).unix();` no FLEX seem to come in anymore.. its been driving me crazy haha