pagermon/pagermon #289: Errors when enabling MessageRepeat Plugin

cwagner17, 2019-05-24:
I have enabled the MessageRepeat plugin and have entered the remote Server URL, API Key and UUID Text. Whenever a message comes in the following error is displayed:

Message failed to deliver. StatusCodeError: 500 - "{\"error\":{\"name\":\"TypeError\",\"message\":\"Cannot read property 'statusCode' of undefined\",\"text\":\"TypeError: Cannot read property 'statusCode' of undefined\"}}"
Retrying in 1000 ms 

There does not seem to be much documentation on this plugin. Is it functional and if so what is causing the above error?

DanrwAU, 2019-05-24:
Can you show us the settings your using?

Confirmed the second pagermon server is working correctly?

cwagner17, 2019-05-24:
I can get my settings, but I wonder if it would be worthwhile to get a description of how to properly set it up.

DanrwAU, 2019-05-24:
[image]

DanrwAU, 2019-05-24:
Also what version are both pagermon servers?

cwagner17, 2019-05-24:
Thanks. It appears I had the URL incorrect as it is connecting and sending messages now. 

But the next issue I am running into is if the remote server is shut down or restarted it crashes the source server. It seems that if MessageRepeater is enabled the remote server needs to be connected or else the source will crash.

Everything is version 0.3.2-beta

Below is the error message that was logged:

[pmon]  2019-05-24 10:25:52  [34mdebug[39m : plugin: MessageRepeat
[pmon]  2019-05-24 10:25:52  [34mdebug[39m : RUNNING PLUGIN!
[pmon]  2019-05-24 10:25:52  [34mdebug[39m : { address: '0283746',
  message: 'test message',
  datetime: '1558707951',
  source: 'TEST',
  pluginData: {} }
[pmon]  2019-05-24 10:25:52  [34mdebug[39m : beforeMessage done
[pmon]  2019-05-24 10:25:52  [31merror[39m : uncaughtException: Cannot read property 'statusCode' of undefined
TypeError: Cannot read property 'statusCode' of undefined
    at Request._callback (/home/pi/pagermon/server/plugins/MessageRepeat.js:50:66)
    at self.callback (/home/pi/pagermon/server/node_modules/request/request.js:185:22)
    at emitOne (events.js:116:13)
    at Request.emit (events.js:211:7)
    at Request.onRequestError (/home/pi/pagermon/server/node_modules/request/request.js:881:8)
    at emitOne (events.js:116:13)
    at ClientRequest.emit (events.js:211:7)
    at Socket.socketErrorListener (_http_client.js:387:9)
    at emitOne (events.js:116:13)
    at Socket.emit (events.js:211:7)
    at emitErrorNT (internal/streams/destroy.js:64:8)
    at _combinedTickCallback (internal/process/next_tick.js:138:11)
    at process._tickDomainCallback (internal/process/next_tick.js:218:9)

DanrwAU, 2019-05-25:
Yeah I did pick that one up ... Something to fix

cwagner17, 2019-05-26:
> Yeah I did pick that one up ... Something to fix

Should we close this ticket and open another one since this got off topic.

marshyonline, 2019-05-26:
Leave this issue open, I'll work on a fix :)
Thanks for bringing it up, I thought I had addressed this is the initial release

DanrwAU, 2019-08-15:
Fixed in #304 and Version 0.3.4

PR #304 (2019-08-14): Fix MessageRepeat error handling
# Description

Fixes error handling of the message repeat plugin.

Previous handling relied on an HTML response, if none was found would crash the application. 

Fixes #289 

## Type of change

Please delete options that are not relevant.

- [x] Bug fix (non-breaking change which fixes an issue)

# How Has This Been Tested?

Tested the following errors
404
401
504 

All throw the correct response

# Checklist:

- [x] My code follows the style guidelines of this project
- [x] I have performed a self-review of my own code
- [x] I have commented my code, particula [... cut]
closed 2019-08-15 by DanrwAU