pagermon/pagermon #290: pagermon client AuthenticationError: Unauthorized

efiten, 2019-06-14:
I installed Pagermon server & client, added an API key in the admin section, added this API key into the config/config.js of the client

When I start the client, i get this error :

> Fail! StatusCodeError: 401 - "{\"error\":{\"name\":\"AuthenticationError\",\"message\":\"Unauthorized\",\"text\":\"AuthenticationError: Unauthorized\"}}"

at that moment the PM2 LOG gives this message : 

> 1|pagermon  | ::ffff:127.0.0.1 - - [14/Jun/2019:10:53:01 +0000] "POST /api/messages HTTP/1.1" 401 108 "-" "PagerMon reader.js"

What am I doing wrong ? are there any other settings to be done ?

Erwin

DanrwAU, 2019-06-14:
That's definetly an API Key issue.

Can you confirm client/config/config.json and server/config/config.json have matching API Keys?

efiten, 2019-06-14:
:(
In some kind of weird way (as you can see in my post, I managed to rename the config file to .JS, and not .JSON...
Now it works, but i'm not able to import my PDW filter.ini file, out of +350 lines, only 15 get imported, the first 10, and another 5 randomly inside the file, any suggestions ?
Or do I need to open another issue for this ?

Erwin

closed 2019-06-14 by efiten