pagermon/pagermon #291: PDW filter.ini import issues

efiten, 2019-06-14:
i'm not able to import my PDW filter.ini file, out of +350 lines, only 15 get imported, the first 10, and another 5 randomly inside the file, any suggestions ?

When I remove the already imported lines, I don't get an error, just 'End of input'

If needed I can send the filters.ini file

DanrwAU, 2019-06-15:
Can you post a sample of ones that do vs don't work? 

I'm not very familiar with the code that does this bit but either way it;'ll likely be required :)

davidmckenzie, 2019-06-16:
The import script is pretty dodgy - hastily thrown together for my particular environment without much regard for other setups. There's a good chance it's looking for something in a different format - or it's just an issue with whitespace or something. Particularly troublesome might be this bit:

```
        if (ol[2].indexOf(" - ") > -1) {
          agency = ol[2].match(/(.*?) - /)[1].trim();
          alias = ol[2].match(/ - (.*?)$/)[1].trim();
        } else {
          agency = '';
          alias = ol[2];
        }
```

It's looking for your labels to be in the format of `Agency - Alias`. You might need to modify this if you have things set up differently. If all your aliases are for the same agency, then you might need to modify the `agency = '';` bit to set a default.

efiten, 2019-06-17:
`2,"0206500","Brandweer LIMBURG Tongeren -OVD-","",0,0,1,4,8,"",1323,04-04-13,17:47:02
2,"0206505","Brandweer LIMBURG Tongeren -Commandant-","",0,0,1,4,8,"",1323,04-04-13,17:47:02
2,"0206506","Brandweer LIMBURG Tongeren -OVD-","",0,0,1,4,8,"",1323,04-04-13,17:47:02
2,"0206518","Brandweer LIMBURG Tongeren -OVD-","",0,0,1,4,8,"",1323,04-04-13,17:47:02
2,"1321302","Brandweer LIMBURG Tongeren -Ploeg 2 (Peloton B)-","",0,0,1,4,8,"",1323,04-04-13,17:47:02
2,"1321301","Brandweer LIMBURG Tongeren -Ploeg 1 (Peloton A)-","",0,0,1,4,8,"",1323,04-04-13,17:47:02
2,"1321303","Brandweer LIMBURG Tongeren -Ploeg 3 (Peloton A)-","",0,0,1,4,8,"",1323,04-04-13,17:47:02
2,"1321304","Brandweer LIMBURG Tongeren -Ploeg 4 (Peloton B)-","",0,0,1,4,8,"",1323,04-04-13,17:47:02
2,"1321305","Brandweer LIMBURG Tongeren -Ploeg 5 (Peloton A)-","",0,0,1,4,8,"",1323,04-04-13,17:47:02
2,"1321306","Brandweer LIMBURG Tongeren -Ploeg 6 (Peloton B)-","",0,0,1,4,8,"",1323,04-04-13,17:47:02
2,"1321307","Brandweer LIMBURG Tongeren -Ploeg 7 (Peloton A)-","",0,0,1,4,8,"",1323,04-04-13,17:47:02
2,"1321309","Brandweer LIMBURG Tongeren -Algemene Oproep)-","",0,0,1,4,8,"",1323,04-04-13,17:47:02
2,"1321310","Brandweer LIMBURG Tongeren -Peloton A-","",0,0,1,4,8,"",1323,04-04-13,17:47:02
2,"1321311","Brandweer LIMBURG Tongeren -Peloton B-","",0,0,1,4,8,"",1323,04-04-13,17:47:02
2,"1321312","Brandweer LIMBURG Tongeren -Peloton A Chauffeurs-","",0,0,1,4,8,"",1323,04-04-13,17:47:02
2,"1321308","Brandweer LIMBURG Tongeren -Ploeg 8 (Peloton B)-","",0,0,1,4,8,"",1323,04-04-13,17:47:02
2,"13213??","Brandweer LIMBURG Tongeren","",0,0,1,4,8,"",1323,04-04-13,17:47:02`

This is a small part of my file, if you want, I can send you the complete file by mail ? just contact me at : erwin.fiten@gmail.com

davidmckenzie, 2019-06-17:
Yeah those won't match the lines I mentioned - you'd need to modify that to suit your format. Also not sure if github just malformed it but looks like that's all on a single line? You might need to `dos2unix` it too.

efiten, 2019-06-17:
I use sublime text editor to produce linux files, so that should not be a problem.
What es the format that migration tool expects ?
I can also alter the filters.ini file to match the format the import tool needs, I think that's easier ?

Erwin

davidmckenzie, 2019-06-17:
See my previous comment - it's looking for "Agency - Alias". You probably just want to change line 117 to include a default agency name. You might also want to change the switch in line 122 to set an icon for the agency.

DanrwAU, 2020-06-03:
I'd suggest converting this to CSV and using the new import function in v0.3.8 and above 😁

closed 2020-06-03 by DanrwAU