pagermon/pagermon #328: Add array/object/collection variable type to plugin confs

davidmckenzie, 2019-09-27:
**Is your feature request related to a problem? Please describe.**
Some plugins may require an array or collection to either take multiple complex variables (such as a list of regex strings), or to control multiple executions (multiple URLs and associated API keys for a webhook plugin, or multiple email addresses and associated names or subjects)

Ref #325 for usecase

**Describe the solution you'd like**
A multirow set similar to that used in the settings page for text matching and the auth keys. There is button for adding new rows and checkboxes for deleting. This would populate either an array of objects, or a collection type object.

The pluginconf definition for this type would include a list of the columns to use, with a limited number of variable types available

**Describe alternatives you've considered**
There can be only one

stubbers, 2019-09-28:
Is there a possibility of having plugin 'instances'. Rather than allowing the plugin to specify multiple URLs allow multiple 'instances' of the plugin somehow?

For example - Rather than having one webhook plugin with an array of destination URLs have multiple instances and allow users to choose them independently per alias

eopo, 2019-09-28:
Ill make an PR and some tests later