pagermon/pagermon #329: Plugin instances

eopo, 2019-09-28:
**Is your feature request related to a problem? Please describe.**
If I wanted to use multiple Pushover Apps (e.g. to reduce the message count for each account and therefore have higher limits) or wanted to use multiple webhooks (See #325), I'd have to copy the plugins code, rename it slightly (So pagermon sees it as a seperate plugin) and create a second config for it.

**Describe the solution you'd like**
I want to create multiple configuration sets for one plugin. From the admin GUI, I want to create new configuration sets, duplicate existing, remove existing, edit existing and give them an order that is respected when running the plugin.
Configuration sets should have a nickname for easier organization and a uuid, that can be referenced elsewhere.

**Describe alternatives you've considered**
Copy the code.

Related to #328, #325

eopo, 2019-09-30:
Rather
```json
"plugins": {
    "plugin_instance1": {
      "plugin": "myFancyPlugin",
      "enable": true,
      "otherOption": "..."
    },
    "plugin_instance2": {
      "plugin": "myFancyPlugin",
      "enable": true,
      "otherOption": "..."
    },
    "otherPlugin": {},
},
```
or
```json
"plugins": {
    "myFancyPlugin": [
      {
        "instance": "plugin_instance1",
        "enable": true,
        "otherOption": "..."
      },
      {
        "instance": "plugin_instance2",
        "enable": true,
        "otherOption": "..."
      }
    ],
    "otherPlugin": {"one could use objects here too, if only one instance is used": ""}
},
```

davidmckenzie, 2019-10-01:
Worth considering how existing configs would be handled - would they need to be converted to a new format on first boot or would the code maintain backwards compatibility.

The second option of your three is probably the most "correct", but with the single instance ones also being an array for consistency.

eopo, 2019-10-01:
I thought about checking wether it’s an array (in this case it’s multiple instances) or an object (in this case it’s just one instance. If a second instance is generated, the first one would be wrapped into the array. 

Alternatively: check config on boot, if any object is detected where array is expected, just wrap it and store