pagermon/pagermon #338: New plugin triggers

eopo, 2019-10-18:
New plugin trigger levels:

Level | Time | Comment
------|-------|---------
before | Before insert | Same as before. Can be used to modify data, before it is inserted
beforePush | After insert, before push | This one is triggered after the data is inserted. Plugins can now insert data themselves, that depends on inserted data (e.G. with foreign keys etc) or could alter the inserted message (for whatever reason one would like to do that)
push | The push itself | Somewhat what after does now - This one can be used for transport plugins, that do not alter the message and only relay it to other services, eg. pushover, mail, telegram, web hook, ... Websockets should use this level, so they can use data extracted by plugins. Additionally plugins should return a success or fail state, that could be stored in the db.
after | after push | This level could alter the message again, but after it has been inserted and pushed.

davidmckenzie, 2019-10-20:
I like it, all makes sense. `afterTransport` could be an interesting one if the `transport` plugins were storing response states, since it could trigger an alternative notification if something fails

eopo, 2019-10-21:
Hmm. How about plugins being able to trigger matching and transport on their own. So Plugin xyz says "nah, don't like it, please rematch". And, like you mentioned: Transport didn't go well, please re-try.

eopo, 2019-10-21:
I ultimately want to implement an advanced user-defined filter system that can say something like:
```
(
    message.ignore = false
    &&
    (
        (
            message.alias.agency
            LIKE
            "AMB"
        )
        OR
        (
            message.plugin.pushover.userkey
            =
            "xyzzy"
        )
    )
=>
run plugin pushover with key xyzzy
)
```