pagermon/pagermon #346: TypeError: Cannot read property '1' of null

dh47, 2019-11-19:
Crashed today with this message.  Running on Raspberry Pi.

```
readline.js:1032
            throw err;
            ^

TypeError: Cannot read property '1' of null
    at Interface.rl.on (/home/pi/pagermon/client/reader.js:104:48)
    at Interface.emit (events.js:189:13)
    at Interface._onLine (readline.js:290:10)
    at Interface._line (readline.js:638:8)
    at Interface._ttyWrite (readline.js:926:16)
    at Socket.onkeypress (readline.js:168:10)
    at Socket.emit (events.js:189:13)
    at emitKeys (internal/readline.js:424:14)
    at emitKeys.next (<anonymous>)
    at Socket.onData (readline.js:1022:36)
Signal caught, exiting!

User cancel, exiting...
```

marshyonline, 2019-11-19:
It looks like the process might have been killed underneath it 
Did the pi happen to run out of memory at the same time?

dh47, 2019-11-19:
Not sure, possible to debug to check if thats the case?

davidmckenzie, 2019-11-19:
```
  } else if (line.indexOf('FLEX: ') > -1) {
    address = line.match(/FLEX:.*?\[(\d*?)\] /)[1].trim();
```

That's the line it died on - looks like the FLEX message received was malformed as the regex returned `null`. That should probably be a softer fail.

Curious what multimon returned for that to fail - is the issue repeatable or once off?

dh47, 2019-11-19:
So far it's only happened once today.  I'll keep an eye on it and report back

dh47, 2019-11-21:
It doesn't seem to be happing as of now, so I'll close this issue if the problem is to happen again in the future I'll reopen and post more information.

dh47, 2019-11-22:
Happened again, same error as last time.

dh47, 2019-12-12:
Reopening this I as have more infomation.

```
readline.js:1032
            throw err;
            ^

TypeError: Cannot read property '1' of null
    at Interface.rl.on (/home/pi/pagermon/client/reader.js:104:48)                                                 
    at Interface.emit (events.js:189:13)
    at Interface._onLine (readline.js:290:10)
    at Interface._line (readline.js:638:8)
    at Interface._ttyWrite (readline.js:926:16)
    at Socket.onkeypress (readline.js:168:10)
    at Socket.emit (events.js:189:13)
    at emitKeys (internal/readline.js:424:14)
    at emitKeys.next (<anonymous>)
    at Socket.onData (readline.js:1022:36)
Signal caught, exiting!

User cancel, exiting...
./reader.sh: line 23:  1952 Done                    rtl_fm -A lut -s 22050 -f 157.950M -                           
      1953 Broken pipe             | multimon-ng -t raw -a POCSAG512 -a POCSAG1200 -a FLEX -f alpha /dev/stdin     
      1954 Aborted                 | node reader.js
```

dh47, 2020-01-08:
...

closed 2020-01-08 by dh47

instantchow, 2021-05-06:
Same error from my setup, I think its not handling funky characters multimon-ng is outputting sometimes. 
```

readline.js:1147
            throw err;
            ^

TypeError: Cannot read property '1' of null
    at Interface.<anonymous> (/home/chow/Projects/pagermon/client/reader.js:119:71)
    at Interface.emit (events.js:314:20)
    at Interface._onLine (readline.js:329:10)
    at Interface._line (readline.js:658:8)
    at Interface._ttyWrite (readline.js:1010:16)
    at Socket.onkeypress (readline.js:205:10)
    at Socket.emit (events.js:314:20)
    at emitKeys (internal/readline/utils.js:335:14)
    at emitKeys.next (<anonymous>)
    at Socket.onData (readline.js:1137:36)

```

irelgrelg, 2021-07-06:
same here, and I was able to capture the 'offending'  message.
My crash happened (at least once) when receiving a follow up message (KFC: C) without receiving the first message (KFC:F)
It tries to delete a frag[address] that doesn't exist.

My problem is though, even though I can read the code, I stupidly enough don't have the ability to write it. So a fix is beyond me... I'll try regardless though and will update here when succesfull.

arcreigh, 2021-11-10:
Can confirm this is a constant problem for me in my environment.

HobbyKonijn, 2021-12-21:
Same issue:
Following messeage caused issue:
FLEX|2021-12-20 16:52:52|1600/2/K/A|13.037|4294941471|ALN|

node:internal/readline/emitKeypressEvents:71
            throw err;
            ^
TypeError: Cannot read properties of null (reading '1')

Looks like no data is behind |ALN| and caused a crash.
Any suggestions how to slove this?

CramShack, 2022-05-29:
Why not just wrap around a try/catch around the message parsing? If a msg is malformed (which they might be - for example, the reader cannot cope with multi-line pager messages) it will simply go into an error log, while allowing the reader to continue without halting.

canuckcam, 2023-05-20:
I'm unfortunately still running into this issue on the latest pagermon if anyone can let me know what I can do to get rid of this error? Thanks!

```
2023-05-20 12:13:44: 000000001: 416710
readline.js:1147
            throw err;
            ^

TypeError: Cannot read property '1' of null
    at Interface.<anonymous> (/home/radio/pagermon/client/reader.js:121:77)
    at Interface.emit (events.js:314:20)
    at Interface._onLine (readline.js:329:10)
    at Interface._line (readline.js:658:8)
    at Interface._ttyWrite (readline.js:1010:16)
    at Socket.onkeypress (readline.js:205:10)
    at Socket.emit (events.js:314:20)
    at emitKeys (internal/readline/utils.js:335:14)
    at emitKeys.next (<anonymous>)
    at Socket.onData (readline.js:1137:36)
Signal caught, exiting!

User cancel, exiting...
```
```
2023-05-20 12:19:15: 001549404: CLR-CONS: Normal to Alarm, CH6 SYSTM ELARM ALARM Ack eqd.
readline.js:1147
            throw err;
            ^

TypeError: Cannot read property '1' of null
    at Interface.<anonymous> (/home/radio/pagermon/client/reader.js:121:77)
    at Interface.emit (events.js:314:20)
    at Interface._onLine (readline.js:329:10)
    at Interface._line (readline.js:658:8)
    at Interface._ttyWrite (readline.js:1010:16)
    at Socket.onkeypress (readline.js:205:10)
    at Socket.emit (events.js:314:20)
    at emitKeys (internal/readline/utils.js:335:14)
    at emitKeys.next (<anonymous>)
    at Socket.onData (readline.js:1137:36)
Signal caught, exiting!

User cancel, exiting...
```

geelongmicrosoldering, 2023-05-20:
replace line 121 of reader.js with the following
`message = line.match(/FLEX[:|].*[|\[][0-9 ]*[|\]] ?...[ |]{0,1}(.*)/)[1].trim();`

I was only given one example of a message that causes this and slightly modified the regex so that [1] is always defined. I actually did that back in 2021 and with the single example provided, it fixed the issue.

You can see here with new regex, the line matches and group 1 is defined, but empty: https://regex101.com/r/S8HMZv/1
This is the old regex, where group 1 is undefined, because the regex doesn't match at all: https://regex101.com/r/BdxbfY/1

There was a PR that involved wrapping statements in try/catch, but i dont think it was finished. Then reader.js was rewritten and I lost track of who was doing what. Anyway, that will solve it.

If you want to prevent reader.sh from stopping when it encounters an error, you can also do this (where ... is your existing config, just add "do" to the start of that line):
```
while true;
do rtl-fm ... | node reader.js;
sleep 10;
done
```
that will restart the reader after a 10 second delay if something causes it to fail

canuckcam, 2023-05-20:
Nailed it with the proper regex pattern. Thank you!