pagermon/pagermon #351: Unable to install pagermon server

SkryptKiddie, 2019-12-10:
After I install the dependencies and clone the repo, cd'ing into the server folder and attempting to run 'sudo npm install' returns this error
npm WARN discord.js@11.5.1 requires a peer of @discordjs/uws@^10.149.0 but none is installed. You must install peer dependencies yourself.

I've tried running 'sudo npm install uws' but that just throws the same error, and gives me a severe security vulnerability error too for the requests module.

DanrwAU, 2019-12-10:
These messages are advisory only. 

Any errors relating to discordjs can be safely ignored.

SkryptKiddie, 2019-12-10:
Well they can't, because it isn't letting me install.

SkryptKiddie, 2019-12-10:
`pi@joshek-pi:~/pagermon/server $ sudo npm install
npm WARN discord.js@11.5.1 requires a peer of @discordjs/uws@^10.149.0 but none is installed. You must install peer dependencies yourself.

audited 2128 packages in 28.936s

2 packages are looking for funding
  run `npm fund` for details

found 1 moderate severity vulnerability
  run `npm audit fix` to fix them, or `npm audit` for details
pi@joshek-pi:~/pagermon/server $ 
`

DanrwAU, 2019-12-10:
That's fine and looks correct. 

Did you try actually running pagermon?

SkryptKiddie, 2019-12-10:
I can't. It won't let me install pagermon at all.

DanrwAU, 2019-12-10:
There is no "install"

As long as you ran npm install once it's done.. 

Run `node app.js` in the server folder and see what it outputs.

SkryptKiddie, 2019-12-10:
Error: Cannot find module '/home/pi/pagermon/server/node_modules/sqlite3/lib/binding/node-v67-linux-arm/node_sqlite3.node'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:668:15)
    at Function.Module._load (internal/modules/cjs/loader.js:591:27)
    at Function._load (/home/pi/pagermon/server/node_modules/pmx/lib/transaction.js:94:21)
    at Module.require (internal/modules/cjs/loader.js:723:19)
    at require (internal/modules/cjs/helpers.js:14:16)
    at Object.<anonymous> (/home/pi/pagermon/server/node_modules/sqlite3/lib/sqlite3.js:4:15)
    at Module._compile (internal/modules/cjs/loader.js:816:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:827:10)
    at Module.load (internal/modules/cjs/loader.js:685:32)
    at Function.Module._load (internal/modules/cjs/loader.js:620:12)
[pmon]  2019-12-10 19:55:12  error : uncaughtException: Cannot find module '/home/pi/pagermon/server/node_modules/sqlite3/lib/binding/node-v67-linux-arm/node_sqlite3.node'
Error: Cannot find module '/home/pi/pagermon/server/node_modules/sqlite3/lib/binding/node-v67-linux-arm/node_sqlite3.node'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:668:15)
    at Function.Module._load (internal/modules/cjs/loader.js:591:27)
    at Function._load (/home/pi/pagermon/server/node_modules/pmx/lib/transaction.js:94:21)
    at Module.require (internal/modules/cjs/loader.js:723:19)
    at require (internal/modules/cjs/helpers.js:14:16)
    at Object.<anonymous> (/home/pi/pagermon/server/node_modules/sqlite3/lib/sqlite3.js:4:15)
    at Module._compile (internal/modules/cjs/loader.js:816:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:827:10)
    at Module.load (internal/modules/cjs/loader.js:685:32)
    at Function.Module._load (internal/modules/cjs/loader.js:620:12)
uncaughtException: Cannot find module '/home/pi/pagermon/server/node_modules/sqlite3/lib/binding/node-v67-linux-arm/node_sqlite3.node'
Error: Cannot find module '/home/pi/pagermon/server/node_modules/sqlite3/lib/binding/node-v67-linux-arm/node_sqlite3.node'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:668:15)
    at Function.Module._load (internal/modules/cjs/loader.js:591:27)
    at Function._load (/home/pi/pagermon/server/node_modules/pmx/lib/transaction.js:94:21)
    at Module.require (internal/modules/cjs/loader.js:723:19)
    at require (internal/modules/cjs/helpers.js:14:16)
    at Object.<anonymous> (/home/pi/pagermon/server/node_modules/sqlite3/lib/sqlite3.js:4:15)
    at Module._compile (internal/modules/cjs/loader.js:816:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:827:10)
    at Module.load (internal/modules/cjs/loader.js:685:32)
    at Function.Module._load (internal/modules/cjs/loader.js:620:12)
[db]  2019-12-10 19:55:12  error : uncaughtException: Cannot find module '/home/pi/pagermon/server/node_modules/sqlite3/lib/binding/node-v67-linux-arm/node_sqlite3.node'
Error: Cannot find module '/home/pi/pagermon/server/node_modules/sqlite3/lib/binding/node-v67-linux-arm/node_sqlite3.node'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:668:15)
    at Function.Module._load (internal/modules/cjs/loader.js:591:27)
    at Function._load (/home/pi/pagermon/server/node_modules/pmx/lib/transaction.js:94:21)
    at Module.require (internal/modules/cjs/loader.js:723:19)
    at require (internal/modules/cjs/helpers.js:14:16)
    at Object.<anonymous> (/home/pi/pagermon/server/node_modules/sqlite3/lib/sqlite3.js:4:15)
    at Module._compile (internal/modules/cjs/loader.js:816:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:827:10)
    at Module.load (internal/modules/cjs/loader.js:685:32)
    at Function.Module._load (internal/modules/cjs/loader.js:620:12)

SkryptKiddie, 2019-12-10:
Wrong button, apologies

DanrwAU, 2019-12-10:
Delete the node_modules folder and re-run npm install then post the logs. 

I've always run into issues with sudoing npm install...

DanrwAU, 2019-12-10:
For reference i just ran it on my windows box and this was the output, all working correctly.

```
c:\Development\pagermon-test\server>npm install
npm WARN deprecated core-js@2.6.11: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
npm WARN deprecated node-uuid@1.4.8: Use uuid module instead
npm WARN deprecated snekfetch@3.6.4: use node-fetch instead

> sqlite3@4.0.9 install c:\Development\pagermon-test\server\node_modules\sqlite3
> node-pre-gyp install --fallback-to-build

node-pre-gyp WARN Using request for node-pre-gyp https download
[sqlite3] Success: "c:\Development\pagermon-test\server\node_modules\sqlite3\lib\binding\node-v64-win32-x64\node_sqlite3.node" is installed via remote

> core-js@2.6.11 postinstall c:\Development\pagermon-test\server\node_modules\core-js
> node -e "try{require('./postinstall')}catch(e){}"

Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfilling JavaScript standard library!

The project needs your help! Please consider supporting of core-js on Open Collective or Patreon:
> https://opencollective.com/core-js
> https://www.patreon.com/zloirock

Also, the author of core-js ( https://github.com/zloirock ) is looking for a good job -)

npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN discord.js@11.5.1 requires a peer of bufferutil@^4.0.0 but none is installed. You must install peer dependencies yourself.
npm WARN discord.js@11.5.1 requires a peer of erlpack@discordapp/erlpack but none is installed. You must install peer dependencies yourself.
npm WARN discord.js@11.5.1 requires a peer of libsodium-wrappers@^0.7.3 but none is installed. You must install peer dependencies yourself.
npm WARN discord.js@11.5.1 requires a peer of node-opus@^0.2.7 but none is installed. You must install peer dependencies yourself.
npm WARN discord.js@11.5.1 requires a peer of opusscript@^0.0.6 but none is installed. You must install peer dependencies yourself.
npm WARN discord.js@11.5.1 requires a peer of sodium@^2.0.3 but none is installed. You must install peer dependencies yourself.
npm WARN discord.js@11.5.1 requires a peer of @discordjs/uws@^10.149.0 but none is installed. You must install peer dependencies yourself.

added 506 packages from 414 contributors and audited 2125 packages in 16.254s
found 1 moderate severity vulnerability
  run `npm audit fix` to fix them, or `npm audit` for details

```

SkryptKiddie, 2019-12-10:
pi@joshek-pi:~/pagermon/server $ sudo npm install

> sqlite3@4.0.9 install /home/pi/pagermon/server/node_modules/sqlite3
> node-pre-gyp install --fallback-to-build

node-pre-gyp WARN Using request for node-pre-gyp https download 
node-pre-gyp WARN Tried to download(403): https://mapbox-node-binary.s3.amazonaws.com/sqlite3/v4.0.9/node-v67-linux-arm.tar.gz 
node-pre-gyp WARN Pre-built binaries not found for sqlite3@4.0.9 and node@11.15.0 (node-v67 ABI, glibc) (falling back to source compile with node-gyp) 
gyp WARN EACCES current user ("pi") does not have permission to access the dev dir "/root/.cache/node-gyp/11.15.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/pagermon/server/node_modules/sqlite3/.node-gyp"
make: Entering directory '/home/pi/pagermon/server/node_modules/sqlite3/build'
make: *** No rule to make target '../.node-gyp/11.15.0/include/node/common.gypi', needed by 'Makefile'.  Stop.
make: Leaving directory '/home/pi/pagermon/server/node_modules/sqlite3/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:193:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:255:12)
gyp ERR! System Linux 4.19.75-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/pi/pagermon/server/node_modules/sqlite3/lib/binding/node-v67-linux-arm/node_sqlite3.node" "--module_name=node_sqlite3" "--module_path=/home/pi/pagermon/server/node_modules/sqlite3/lib/binding/node-v67-linux-arm" "--napi_version=4" "--node_abi_napi=napi" "--napi_build_version=0" "--node_napi_label=node-v67"
gyp ERR! cwd /home/pi/pagermon/server/node_modules/sqlite3
gyp ERR! node -v v11.15.0
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/node /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/pagermon/server/node_modules/sqlite3/lib/binding/node-v67-linux-arm/node_sqlite3.node --module_name=node_sqlite3 --module_path=/home/pi/pagermon/server/node_modules/sqlite3/lib/binding/node-v67-linux-arm --napi_version=4 --node_abi_napi=napi --napi_build_version=0 --node_napi_label=node-v67' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/pagermon/server/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:193:13)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:999:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:266:5)
node-pre-gyp ERR! System Linux 4.19.75-v7+
node-pre-gyp ERR! command "/usr/bin/node" "/home/pi/pagermon/server/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/pagermon/server/node_modules/sqlite3
node-pre-gyp ERR! node -v v11.15.0
node-pre-gyp ERR! node-pre-gyp -v v0.11.0
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/node /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/pagermon/server/node_modules/sqlite3/lib/binding/node-v67-linux-arm/node_sqlite3.node --module_name=node_sqlite3 --module_path=/home/pi/pagermon/server/node_modules/sqlite3/lib/binding/node-v67-linux-arm --napi_version=4 --node_abi_napi=napi --napi_build_version=0 --node_napi_label=node-v67' (1)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! sqlite3@4.0.9 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the sqlite3@4.0.9 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-12-10T12_06_00_572Z-debug.log
pi@joshek-pi:~/pagermon/server $

DanrwAU, 2019-12-10:
Its complaining  about Permissions during the sqlite install... does it do the same thing if not sudod?

SkryptKiddie, 2019-12-10:
I'll give that a try

SkryptKiddie, 2019-12-10:
Seems stuck at this point

make: Entering directory '/home/pi/pagermon/server/node_modules/sqlite3/build'
  ACTION deps_sqlite3_gyp_action_before_build_target_unpack_sqlite_dep Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c
  TOUCH Release/obj.target/deps/action_before_build.stamp
  CC(target) Release/obj.target/sqlite3/gen/sqlite-autoconf-3280000/sqlite3.o
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c: In function ‘strftimeFunc’:
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c:14137:27: warning: cast between incompatible function types from ‘int (*)(void *)’ to ‘void (*)(void *)’ [-Wcast-function-type]
 #define SQLITE_DYNAMIC   ((sqlite3_destructor_type)sqlite3MallocSize)
                           ^
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c:22052:52: note: in expansion of macro ‘SQLITE_DYNAMIC’
                       z==zBuf ? SQLITE_TRANSIENT : SQLITE_DYNAMIC);
                                                    ^~~~~~~~~~~~~~
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c: In function ‘sqlite3ErrorWithMsg’:
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c:14137:27: warning: cast between incompatible function types from ‘int (*)(void *)’ to ‘void (*)(void *)’ [-Wcast-function-type]
 #define SQLITE_DYNAMIC   ((sqlite3_destructor_type)sqlite3MallocSize)
                           ^
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c:30393:54: note: in expansion of macro ‘SQLITE_DYNAMIC’
     sqlite3ValueSetStr(db->pErr, -1, z, SQLITE_UTF8, SQLITE_DYNAMIC);
                                                      ^~~~~~~~~~~~~~
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c: In function ‘sqlite3VdbeMemSetStr’:
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c:14137:27: warning: cast between incompatible function types from ‘int (*)(void *)’ to ‘void (*)(void *)’ [-Wcast-function-type]
 #define SQLITE_DYNAMIC   ((sqlite3_destructor_type)sqlite3MallocSize)
                           ^
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c:75499:15: note: in expansion of macro ‘SQLITE_DYNAMIC’
     if( xDel==SQLITE_DYNAMIC ){
               ^~~~~~~~~~~~~~
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c: In function ‘valueFromExpr’:
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c:14137:27: warning: cast between incompatible function types from ‘int (*)(void *)’ to ‘void (*)(void *)’ [-Wcast-function-type]
 #define SQLITE_DYNAMIC   ((sqlite3_destructor_type)sqlite3MallocSize)
                           ^
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c:75905:55: note: in expansion of macro ‘SQLITE_DYNAMIC’
       sqlite3ValueSetStr(pVal, -1, zVal, SQLITE_UTF8, SQLITE_DYNAMIC);
                                                       ^~~~~~~~~~~~~~
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c:14137:27: warning: cast between incompatible function types from ‘int (*)(void *)’ to ‘void (*)(void *)’ [-Wcast-function-type]
 #define SQLITE_DYNAMIC   ((sqlite3_destructor_type)sqlite3MallocSize)
                           ^
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c:75948:29: note: in expansion of macro ‘SQLITE_DYNAMIC’
                          0, SQLITE_DYNAMIC);
                             ^~~~~~~~~~~~~~
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c: In function ‘sqlite3VdbeExec’:
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c:91000:38: warning: cast between incompatible function types from ‘int (*)(u32,  void *, void *, void *)’ {aka ‘int (*)(unsigned int,  void *, void *, void *)’} to ‘void (*)(void *, const char *)’ [-Wcast-function-type]
       void (*x)(void*,const char*) = (void(*)(void*,const char*))db->xTrace;
                                      ^
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c: In function ‘printfFunc’:
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c:14137:27: warning: cast between incompatible function types from ‘int (*)(void *)’ to ‘void (*)(void *)’ [-Wcast-function-type]
 #define SQLITE_DYNAMIC   ((sqlite3_destructor_type)sqlite3MallocSize)
                           ^
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c:113251:25: note: in expansion of macro ‘SQLITE_DYNAMIC’
                         SQLITE_DYNAMIC);
                         ^~~~~~~~~~~~~~
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c: In function ‘generateColumnNames’:
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c:14137:27: warning: cast between incompatible function types from ‘int (*)(void *)’ to ‘void (*)(void *)’ [-Wcast-function-type]
 #define SQLITE_DYNAMIC   ((sqlite3_destructor_type)sqlite3MallocSize)
                           ^
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c:126515:58: note: in expansion of macro ‘SQLITE_DYNAMIC’
         sqlite3VdbeSetColName(v, i, COLNAME_NAME, zName, SQLITE_DYNAMIC);
                                                          ^~~~~~~~~~~~~~
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c:14137:27: warning: cast between incompatible function types from ‘int (*)(void *)’ to ‘void (*)(void *)’ [-Wcast-function-type]
 #define SQLITE_DYNAMIC   ((sqlite3_destructor_type)sqlite3MallocSize)
                           ^
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c:126522:52: note: in expansion of macro ‘SQLITE_DYNAMIC’
       sqlite3VdbeSetColName(v, i, COLNAME_NAME, z, SQLITE_DYNAMIC);
                                                    ^~~~~~~~~~~~~~
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c: In function ‘sqlite3InvokeBusyHandler’:
[... cut ...]
       int_as_real: /* fall through to real */;
       ^~~~~~~~~~~
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c:179183:5: note: here
     case JSON_REAL: {
     ^~~~
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c: In function ‘jsonEachColumn’:
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c:180825:9: warning: this statement may fall through [-Wimplicit-fallthrough=]
       if( p->bRecursive ){
         ^
Release/obj/gen/sqlite-autoconf-3280000/sqlite3.c:180835:5: note: here
     default: {
     ^~~~~~~

SkryptKiddie, 2019-12-10:
It's printing more in the terminal, still loads of warnings, etc.

SkryptKiddie, 2019-12-10:
It's finished installing, I think it's worked. Do I just run 'node app.js' now?

SkryptKiddie, 2019-12-10:
All good now, managed to start the server. Thanks for your help.

closed 2019-12-10 by SkryptKiddie