pagermon/pagermon #352: Add ability to update malformed messages

DanrwAU, 2019-12-17:
**Is your feature request related to a problem? Please describe.**
When using multiple receivers, it would be nice to have Pagermon replace duplicates that may be malformed with the correct version if another receiver has it. 

**Describe the solution you'd like**
Some form of matching to replace messages if received within 5 (?) seconds or so of another message with the same address and similar content. 

**Additional context**
Example
[image]

Bangs69, 2020-09-20:
Hi DanrwAU,
Is there an easy way to setup the groups (SES/RFS/etc..) or did you just slowly add them as identified?
I have just gotten the scanner function to work with multiple frequencies on the one dongle and now my feed has become more difficult to read without some way to identify which service the pages came from. 
Would be great to have it looking like yours!

DanrwAU, 2020-09-20:
Offtopic comment removed, capcode sharing is not in scope of the project.

Bangs69, 2020-09-20:
Sorry about that mate. -- Sent from my Android phone with mail.com Mail. Please excuse my brevity.