pagermon/pagermon #353: Add duplicate capcode filter between networks

melbournefan, 2020-01-07:
When capcodes are added, provide a option in the settings that 'binds' a capcode to one receiver (network), I'm starting to find that some capcodes are conflicting between networks

davidmckenzie, 2020-01-07:
Possibly dupe of #141 - I'm guessing you're thinking to separate them out based on the source field?

melbournefan, 2020-01-08:
Yup, one of the CFA capcodes on the EAS network is conflicting with a telematics capcode on Vodafone that I want to block

DanrwAU, 2020-01-08:
I filter this on the receiver side. 

Example from my reader.js 

`if ((address.length > 2 && message) && ((trimMessage.match(/E\d{11}/g)) || (trimMessage.match(/N\d{11}/g)) || (trimMessage.match(/J\d{11}/g)) || (trimMessage.match('Part 2 of 2')) || (trimMessage.match('Branch for cover')))) {    `

closed 2020-01-18 by melbournefan