pagermon/pagermon #355: Incorrect Pushover Key crashes application

DanrwAU, 2020-01-27:
**Describe the bug**
Incorrect API Key in pushover causes application crash

**To Reproduce**
Enable Pushover on alias with incorrect API Key. 

**Expected behavior**
Application Doesn't crash

**Logs**
```
3|pagermon-test  | Error: user identifier is not a valid user, group, or subscribed user key
3|pagermon-test  |     at Pushover.errors (/home/daniel/pagermon-test/server/node_modules/pushover-notifications/lib/pushover.js:145:13)
3|pagermon-test  |     at IncomingMessage.<anonymous> (/home/daniel/pagermon-test/server/node_modules/pushover-notifications/lib/pushover.js:249:12)
3|pagermon-test  |     at IncomingMessage.emit (events.js:194:15)
3|pagermon-test  |     at IncomingMessage.EventEmitter.emit (domain.js:441:20)
3|pagermon-test  |     at endReadableNT (_stream_readable.js:1125:12)
3|pagermon-test  |     at process._tickCallback (internal/process/next_tick.js:63:19)
3|pagermon-test  | [pmon]  2020-01-27 08:16:40  error : uncaughtException: user identifier is not a valid user, group, or subscribed user key
3|pagermon-test  | Error: user identifier is not a valid user, group, or subscribed user key
3|pagermon-test  |     at Pushover.errors (/home/daniel/pagermon-test/server/node_modules/pushover-notifications/lib/pushover.js:145:13)
3|pagermon-test  |     at IncomingMessage.<anonymous> (/home/daniel/pagermon-test/server/node_modules/pushover-notifications/lib/pushover.js:249:12)
3|pagermon-test  |     at IncomingMessage.emit (events.js:194:15)
3|pagermon-test  |     at IncomingMessage.EventEmitter.emit (domain.js:441:20)
3|pagermon-test  |     at endReadableNT (_stream_readable.js:1125:12)
3|pagermon-test  |     at process._tickCallback (internal/process/next_tick.js:63:19)
3|pagermon-test  | uncaughtException: user identifier is not a valid user, group, or subscribed user key
3|pagermon-test  | Error: user identifier is not a valid user, group, or subscribed user key
3|pagermon-test  |     at Pushover.errors (/home/daniel/pagermon-test/server/node_modules/pushover-notifications/lib/pushover.js:145:13)
3|pagermon-test  |     at IncomingMessage.<anonymous> (/home/daniel/pagermon-test/server/node_modules/pushover-notifications/lib/pushover.js:249:12)
3|pagermon-test  |     at IncomingMessage.emit (events.js:194:15)
3|pagermon-test  |     at IncomingMessage.EventEmitter.emit (domain.js:441:20)
3|pagermon-test  |     at endReadableNT (_stream_readable.js:1125:12)
3|pagermon-test  |     at process._tickCallback (internal/process/next_tick.js:63:19)
3|pagermon-test  | [db]  2020-01-27 08:16:40  error : uncaughtException: user identifier is not a valid user, group, or subscribed user key
3|pagermon-test  | Error: user identifier is not a valid user, group, or subscribed user key
3|pagermon-test  |     at Pushover.errors (/home/daniel/pagermon-test/server/node_modules/pushover-notifications/lib/pushover.js:145:13)
3|pagermon-test  |     at IncomingMessage.<anonymous> (/home/daniel/pagermon-test/server/node_modules/pushover-notifications/lib/pushover.js:249:12)
3|pagermon-test  |     at IncomingMessage.emit (events.js:194:15)
3|pagermon-test  |     at IncomingMessage.EventEmitter.emit (domain.js:441:20)
3|pagermon-test  |     at endReadableNT (_stream_readable.js:1125:12)
3|pagermon-test  |     at process._tickCallback (internal/process/next_tick.js:63:19)
```
**Environment information:**
 - Version: 0.3.5-beta and 0.3.6-beta

**Additional context**
Issue was fixed in a previous version, regression somewhere along the way, possibly conversion to plugin system?

DanrwAU, 2020-01-27:
@davidmckenzie any ideas on this one, code looks identical to the working one in 0.1.8 so i think it's something with the plugin system but i don't know it that well. 😢

davidmckenzie, 2020-01-27:
Did we run with different pinned version of pushover lib then? Maybe bug introduced upstream. The library itself seems to be giving the uncaught exception, though maybe we just need to put a try/catch in the plugin

DanrwAU, 2020-01-27:
Uh good point, i think i may have set it to "latest" when i implemented it. 

I recently did an NPM install on all my Prod instances anyone still running an old version and can test @pagermon/collaborators ?

DanrwAU, 2020-03-19:
Looks like this was fixed in v.1.2.2 upstream

https://github.com/qbit/node-pushover/commit/2f2fcb6a77778c301a136c660845c71231600ba7

I'll upgrade a test instance today and check it out

DanrwAU, 2020-03-19:
Still happening dammit 

different tho 
````
3|pagermon-test  | Error: user identifier is not a valid user, group, or subscribed user key
3|pagermon-test  |     at Pushover.errors (/home/daniel/pagermon-test/server/node_modules/pushover-notifications/lib/pushover.js:146:13)
3|pagermon-test  |     at IncomingMessage.<anonymous> (/home/daniel/pagermon-test/server/node_modules/pushover-notifications/lib/pushover.js:250:12)
3|pagermon-test  |     at IncomingMessage.emit (events.js:194:15)
3|pagermon-test  |     at IncomingMessage.EventEmitter.emit (domain.js:441:20)
3|pagermon-test  |     at endReadableNT (_stream_readable.js:1125:12)
3|pagermon-test  |     at process._tickCallback (internal/process/next_tick.js:63:19)
3|pagermon-test  | [pmon]  2020-03-19 00:24:28  error : uncaughtException: user identifier is not a valid user, group, or subscribed user key
3|pagermon-test  | Error: user identifier is not a valid user, group, or subscribed user key
3|pagermon-test  |     at Pushover.errors (/home/daniel/pagermon-test/server/node_modules/pushover-notifications/lib/pushover.js:146:13)
3|pagermon-test  |     at IncomingMessage.<anonymous> (/home/daniel/pagermon-test/server/node_modules/pushover-notifications/lib/pushover.js:250:12)
3|pagermon-test  |     at IncomingMessage.emit (events.js:194:15)
3|pagermon-test  |     at IncomingMessage.EventEmitter.emit (domain.js:441:20)
3|pagermon-test  |     at endReadableNT (_stream_readable.js:1125:12)
3|pagermon-test  |     at process._tickCallback (internal/process/next_tick.js:63:19)
3|pagermon-test  | uncaughtException: user identifier is not a valid user, group, or subscribed user key
3|pagermon-test  | Error: user identifier is not a valid user, group, or subscribed user key
3|pagermon-test  |     at Pushover.errors (/home/daniel/pagermon-test/server/node_modules/pushover-notifications/lib/pushover.js:146:13)
3|pagermon-test  |     at IncomingMessage.<anonymous> (/home/daniel/pagermon-test/server/node_modules/pushover-notifications/lib/pushover.js:250:12)
3|pagermon-test  |     at IncomingMessage.emit (events.js:194:15)
3|pagermon-test  |     at IncomingMessage.EventEmitter.emit (domain.js:441:20)
3|pagermon-test  |     at endReadableNT (_stream_readable.js:1125:12)
3|pagermon-test  |     at process._tickCallback (internal/process/next_tick.js:63:19)
3|pagermon-test  | [db]  2020-03-19 00:24:28  error : uncaughtException: user identifier is not a valid user, group, or subscribed user key
3|pagermon-test  | Error: user identifier is not a valid user, group, or subscribed user key
3|pagermon-test  |     at Pushover.errors (/home/daniel/pagermon-test/server/node_modules/pushover-notifications/lib/pushover.js:146:13)
3|pagermon-test  |     at IncomingMessage.<anonymous> (/home/daniel/pagermon-test/server/node_modules/pushover-notifications/lib/pushover.js:250:12)
3|pagermon-test  |     at IncomingMessage.emit (events.js:194:15)
3|pagermon-test  |     at IncomingMessage.EventEmitter.emit (domain.js:441:20)
3|pagermon-test  |     at endReadableNT (_stream_readable.js:1125:12)
3|pagermon-test  |     at process._tickCallback (internal/process/next_tick.js:63:19)
````

DanrwAU, 2020-04-19:
Opened an issue with node-pushover

SloCompTech, 2020-04-19:
Maybe check if api key is valid by sending test notification when key is set ?

DanrwAU, 2020-04-20:
> Maybe check if api key is valid by sending test notification when key is set ?

That would still crash the app, we could wrap it in a try/catch but that seems silly when error handling is already supposedly in the module