pagermon/pagermon #356: Adress link to google

112bzo, 2020-01-28:
is it possible to link addresses to google maps?

eopo, 2020-01-28:
Not out of the box, since you first have to find out which part of the message is the address. I’m currently working on the framing that implements geo-data for messages, but the parsing of the messages is up to you since the format differs from network to network and can even differ inside a single network. We can’t provide an out-of-the-Box „one size fits all“ solution there

all3kcis, 2020-03-26:
You can look at https://github.com/all3kcis/pagermon-SMTPA3K/

HobbyKonijn, 2020-05-16:
16-05-2020 | 13:44 | RTL | 000923130 | AMBU | ambulance 08-130 - AMBU - Regionaal - Gelderland Zuid |A1 Oude Zuiderlingedijk 4147BP Asperen 58907

FLEX
000923130  = Capcode
AMBU = dicipline
ambulance 08-130 - AMBU - Regionaal - Gelderland Zuid = Alias

A1 Oude Zuiderlingedijk 4147BP Asperen 58907 = broadcasted message via FLEX system
->
A1 is prio setting dicipline (A1 / A2 is normally prio for Ambu) could be also P 1 / P 2 for fire brigade
Oude Zuiderlingedijk = address
4147BP = zipcode (I think 112bzo wants on this 6 digit code a google search)
Asperen = City 
58907 = internal case number of dicipline)

I would guess you want to use "4147BP Asperen" for a google search and get a map like
https://www.google.com/maps/search/4147BP+Asperen/

This means ####AB+City need to be taken out of the message.

112bzo, 2020-06-20:
Thats true HobbyKonijn.

will-caruana, 2020-08-27:
It would be interesting to pass phone numbers (find them using regex) to something like opencnam.com then uses the address information from that to pin onto Google maps. Google maps also has a comment area where the most recent pages for that location could be stored. (I am not sure how useful this is outside the US)