pagermon/pagermon #357: Flex no Longer Being Sent to Pagermon Server

dh47, 2020-02-26:
It seems multimon-ng has changed the way Flex is phased.

[image]

This is what Flex looks like now just via multimon-ng.

`FLEX|2020-02-26 06:46:15|1600/2/K/A|11.081|001234567|ALN|This is a test periodic page. 95 19:46`

A few people on the discord has advised that the format expected is different in the reader.js file.

Anyone else can chuck in there 2cents, please?  Ta.

dh47, 2020-02-28:
More raw samples from multimon-ng.  @marshyonline 
```
FLEX|2020-02-28 04:40:08|1600/2/K/A|10.014|001234567|ALN|This is a test periodic page. 21 17:40
FLEX|2020-02-28 04:41:56|1600/2/K/A|10.071|000998866|ALN|ON DUTY RFO NTH CANTY - Unit MOTU9711 [ SITREP : ] FIRE IS ABOUT 7 HECTARES IN SIZE - HELO REQUIRED - SOUTHCOMS
FLEX|2020-02-28 04:42:00|1600/2/K/A|10.073|000998797|ALN|ON DUTY RFO NTH CANTY - Unit MOTU9711 [ SITREP : ] FIRE IS ABOUT 7 HECTARES IN SIZE - HELO REQUIRED - SOUTHCOMS
FLEX|2020-02-28 04:42:10|1600/2/K/A|10.079|001234567|ALN|This is a test periodic page. 23 17:42
FLEX|2020-02-28 04:42:58|1600/2/K/A|10.104|000999849|ALN|(NGAM6811) STRU TARURUTANGI NEW PLYMOUTH DISTRICT. (XStr AWAI RD/HURSTHOUSE RD) .SMOKE FROM SHED AT INTERSECTION MAKE TANKERS 1. #F2958010
FLEX|2020-02-28 04:43:10|1600/2/K/A|10.111|001234567|ALN|This is a test periodic page. 24 17:43
```

takenalias, 2020-02-28:
I think your issue is with multimon. This is my output from multimon. 

FLEX: 2020-02-28 09:34:50 1600/2/K/A 08.090 [001929745] ALN LW4 PR: 4 - : @ST HILARIONS RESIDENTIAL CARE AND NURSING HOME - SEATON SEATON 104 D 9 D01235 Disp: 19:11 Psychiatri
FLEX: 2020-02-28 09:35:29 1600/2/K/A 08.111 [001925578] ALN F112 PR: 3 - : @RFDS: FXN X RDHL ETA 2000 ADELAIDE AIRPORT 128 H 6 D01236 Disp: 20:05 TRANSFERFLEX: 2020-02-28 09:35:29 1600/2/K/A 08.111 [001929672] ALN PFD142 PR: 3 - BLAKEVIEW 42 N 15 D01237 Disp: 20:05 Psychiatri
FLEX: 2020-02-28 09:35:34 1600/2/K/A 08.114 [001900159] ALN mhs-watchdog 200228200529

dh47, 2020-02-28:
Is your build of multimon lastest?

dh47, 2020-02-28:
I'm running multimon-ng 1.1.8.

takenalias, 2020-02-28:
No

takenalias, 2020-02-28:
Im running 1.1.7 I think they have been working on flex decoding. So might be worth checking the change logs.

dh47, 2020-02-28:
Only thing that's in the changelog release.
> FLEX: Numeric messages now won't have spaces stripped anymore.

dh47, 2020-02-28:
Rolling back to 1.1.7 fixed this issue.  I suggest people manually download or git pull the 1.1.7 build for now until the Pagermon team update to the new format multimon-ng uses. `b9bd39a`

DanrwAU, 2020-02-28:
Caused by https://github.com/EliasOenal/multimon-ng/issues/123

dh47, 2020-03-16:
Just wondering if any progress or quick fixes can be made in reguards to this problem?

davidmckenzie, 2020-03-19:
Will take a look this weekend, it'll need some regex mangling for it to match both old and new formats. Those grouped messages will be a bigger problem to solve. For now as quick fix we'll just ignore those and match on the first capcode in the list.

I have no FLEX networks around here so if anyone wanted to tackle parsing the arrayed/grouped capcodes it would be much appreciated

dh47, 2020-03-19:
Let me know what you need, Flex is a huge part of our network here.

dh47, 2020-03-22:
Cool, I'll take a look.

dh47, 2020-03-22:
Looking good here.  @davidmckenzie 

[image]

AndyRPH, 2020-03-22:
Dh47, what are you doing to check that it worked?  I'm having the same issue with my FLEX messages not showing in the server but when I run the command that is in reader.sh to a text file the messages are there. 

I loaded in the revised reader.js  and restarted pagermon but still don't get any in the website.  They do appear if I manually push a correctly formatted message to | node reader.js  so all the internals of pagermon are functioning.

AndyRPH, 2020-03-22:
Nevermind, I must have a different issue.  I snipped a line from a text capture of multimon-ng's output into reader.js and it showed up in the server.  Yet nothing normally is populating the server.  But since my manual entry processes  nicely that may help validate this change. 

My area is almost or completely using FLEX as the protocol. 

echo "FLEX: 2020-03-22 20:11:10 3200/4/A 02.110 [4294949098] NUM 66317[94]" | node reader.js
2020-03-22 17:01:27: 4294949098: 66317[94]
Input died!

And it shows in the website, but not when those lines are fed in from running the process via pm2 start process.json.  

I'll take my issue to discord so I don't muddy this potential correction up!

marshyonline, 2020-03-22:
Can you run multimon without pushing the output into the reader and send us the dump? 
We are going to need some of the raw multimon output to debug this.

marshyonline, 2020-03-22:
Ah, no problems.
Your reply didn't load before I posted mine

marshyonline, 2020-03-26:
This is still an issue?

Oojimentis, 2020-04-10:
Any updates on this? Thanks

dh47, 2020-04-10:
I think it's in the next update.

davidmckenzie, 2020-04-11:
Yep you can pull down the `v0.3.7` branch if you want it now :)

dh47, 2020-04-20:
Keeping this open so people can see.  And will close again when 0.3.7 is released with this fix.

DanrwAU, 2020-04-22:
Finally closing for good!

PR #358 (2020-03-22): fix regex maybe
# Description

Should make the regex work with both old and new formats. My testing was limited to throwing a few test cases through it so I have no idea how it will perform in the real world. There's a high probability this will break things for FLEX users

Fixes #357

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)
closed 2020-04-22 by DanrwAU