pagermon/pagermon #359: Fire rescue call outs are know longer logged

mitchrockwell, 2020-03-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

DanrwAU, 2020-03-22:
Sorry we're gunna need alot more info than this.

What Paging Technology. What system. Anything!

marshyonline, 2020-03-26:
Please just on discord or provide more information

closed 2020-03-26 by marshyonline