pagermon/pagermon #361: WebSocket not working on new messages

all3kcis, 2020-03-24:
See : https://github.com/pagermon/pagermon/blob/2f24061fd6f45dd01bfb0dce7f226dcccbd3e409/server/routes/api.js#L712-L756

This portion of code does not work inside the plugin callback.
Does it need to be in this callback? Can't it be output?
I think it should be out of the callback, as the message is already stored in the database, it can be sent to the socket.io

davidmckenzie, 2020-03-24:
It's in the callback to force it to be synchronous - you're right though, probably doesn't need to be for the `after` plugins. Curious why it would be failing though, broken plugin?

all3kcis, 2020-03-24:
I don't know, but I realized that it worked well out of the plugin callback :)

marshyonline, 2020-03-26:
I'm unable to replicate this issue in the current master branch.
Do you have any local plugins that could be causing this issue?

all3kcis, 2020-03-26:
Maybe, I was in development on https://github.com/pagermon/pagermon/pull/283
Now it seems to be working properly, really sorry.
It can be closed.

closed 2020-04-05 by DanrwAU