pagermon/pagermon #362: Search sometimes redirects to first page when next page is loaded

melbournefan, 2020-04-04:
**Describe the bug**
When a page at the bottom is pressed, the page automatically redirects to the first page of that search, even tho you've clicked on Page 10 (for example)

**To Reproduce**
Steps to reproduce the behavior:
1. Enter a search item in the search bar at the bottom
2. Press Enter
3. Scroll to the bottom and click on a random page number, tho this doesn't always happen
4. It'll appear to load the page you wanted, tho it'll redirect to Page 1, even tho the URL bar says the correct page

**Expected behavior**
The page you've clicked on isn't supposed to redirect to page 1

**Environment information:**
 - OS: Server instance running Ubuntu 16.04.6LTS, Pagermon version 0.3.6, client running Windows 10 on Firefox 74.0

DanrwAU, 2020-04-05:
Any errors in the server side logs? Sounds like the app may be crashing and restarting. 

I haven't been able to reproduce... what DB backend?

melbournefan, 2020-04-12:
Hi,

I got the followingerror when it redirected me `2|app  | (node:1443) Warning: No such label 'sql' for console.timeEnd()` when it redirected me, also using SQLite for the backend.

Also got `2|app      | (node:1443) Warning: No such label 'init' for console.timeEnd()` and `2|app      | (node:1468) Warning: No such label 'updateMap' for console.timeEnd(` as well as the 'sql' label error when I ran `pm2 logs app`

Thanks

davidmckenzie, 2020-04-12:
Weird... any particular search strings or does it happen with all searches? The `updateMap` timeEnd only happens in capcode update routes so double wtf.

melbournefan, 2020-04-13:
It basically happens when I search for something that's more than 5 pages, tho note that I have then filtered capcodes which would then generate the `updateMap` error, it's just there when I first load the logs.

DanrwAU, 2020-04-13:
I suspect your actually running out of memory and the app is restarting.

Try setting `"max_memory_restart": "100M",` to "max_memory_restart": "1G", in your process.json

You'll have to remove it from PM2 and re-add for it to take affect

melbournefan, 2020-04-13:
Added that to the file then deleted / re added it to PM2, still happening. I did look at htop and whilst I saw some spikes in CPU usage, memory was under 2gb out of 4gb for all the processes.
I suspect that the CPU usage is causing pagermon to 'throttle' of sorts, nothing about a restart either in the logs.

DanrwAU, 2020-04-13:
Nope... CPU can be pegged at 100% for over 5 minutes on mine when doing an alias refresh, doesn't usually cause any issues.

Set your log level to debug and post the output of the knex logs from the query.

davidmckenzie, 2020-04-13:
Long shot but you're not using old version of Node.js are you? >8 is required from memory

DanrwAU, 2020-06-03:
I've still not been able to reproduce this one..... Is PM2 showing a restart of the application?

closed 2020-08-10 by melbournefan