pagermon/pagermon #363: Help to get running

kevtheskin, 2020-04-04:
Hello there,
I have trying to get pagermon running for ages and no success. I have got the page up on 127.0.0.1:3000 then i run  rtl_fm -f 153.350M -s 22050 -g37 | multimon-ng -q  -e -t raw -a POCSAG512 -a POCSAG1200 -a POCSAG2400 -a Flex -a SCOPE -f alpha /dev/stdin | node reader.js . Nothing happens on webpage apart from a wheel turning in middle of page. Also multimon crashes. Please help im fairly new to linux.

Cheers Kev

SloCompTech, 2020-04-04:
Hi,
for detailed help would be nice to put error trace if available.
In general how to setup pagermon client:
- install libraries required (rtl_fm, multimon-ng, nodejs)
- blacklist default rtlsdr driver
- reboot
- in config dir copy default.json to config.json
- edit config.json add URL and API KEY (configure on server)
- run reader.sh in rc.local or somewhere similar

[Read this](https://github.com/pagermon/pagermon/tree/master/client)

kevtheskin, 2020-04-05:
Hello there, 
Followed your instructions I think multimon-ng crashes and webpage has a spinning wheel in middle of page.

/home/radio/pagermon/client/node_modules/nconf/lib/nconf/stores/file.js:160
    throw new Error("Error parsing your configuration file: [" + this.file + ']: ' + ex.message);
    ^

Error: Error parsing your configuration file: [./config/config.json]: Unexpected number in JSON at position 12
    at exports.File.File.loadSync (/home/radio/pagermon/client/node_modules/nconf/lib/nconf/stores/file.js:160:11)
    at exports.Provider.Provider.add (/home/radio/pagermon/client/node_modules/nconf/lib/nconf/provider.js:137:23)
    at exports.Provider.Provider.file (/home/radio/pagermon/client/node_modules/nconf/lib/nconf/provider.js:63:15)
    at Object.<anonymous> (/home/radio/pagermon/client/reader.js:26:11)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Function.Module.runMain (module.js:693:10)

SloCompTech, 2020-04-05:
This error trace indicates, that you have error in your client config file (`config.json`)

kevtheskin, 2020-04-05:
Hello again , How do I fix this. My linux skills are still in the early learning stage. Thank you for your time.
Kev

SloCompTech, 2020-04-05:
Hi,
from log above do following:
``` bash
cd /home/radio/pagermon/client/config
cp default.json config.json
nano config.json # Edit config file
```
In configuration file change:
- *identifier* to one configured on pagermon server
- *apikey* to one configured for specified identifier

kevtheskin, 2020-04-05:
> 
> 
> Hi,
> from log above do following:
> 
> ```shell
> cd /home/radio/pagermon/client/config
> cp default.json config.json
> nano config.json # Edit config file
> ```
> 
> In configuration file change:
> 
>     * _identifier_ to one configured on pagermon server
> 
>     * _apikey_ to one configured for specified identifier

How do I enter the key. I thought I had already did this. I will post the contents of this file tomorrow UK time. Thank you for you time . Cheers Kev

DanrwAU, 2020-04-05:
I'd suggest following the tutorials on the wiki and in the readme, you seem to have missed some steps which are clearly labelled there. 

If you still need help, stop by #support in the discord

closed 2020-04-05 by DanrwAU

kevtheskin, 2020-04-06:
Hi can you give me the link for the discord please?

Kev

SloCompTech, 2020-04-06:
https://discord.gg/3VK7gSD