pagermon/pagermon #402: Delete button on AliasDetail screen doesn't work

DanrwAU, 2020-05-07:
**Describe the bug**
The delete button on the AliasDetail screen doesn't work in the latest version - seems to be independent of the DB backend. 

**To Reproduce**
Open an alias, try to delete, receive spinning wheel of death.

Logs show the request is never submitted to the API

**Expected behavior**
Delete request is submitted to the api and alias is deleted.

DanrwAU, 2020-05-07:
Looks like line 463 in api.js never actually fires it does everything before it, but the api never gets the request.

I'm confused.....

```
        Api.AliasDetail.delete({id: $routeParams.id}, $scope.alias).$promise.then(function (response) {
```

DanrwAU, 2020-05-07:
Changing the above line to Api.AliasDetail.save does save the capcode, so there is nothing wrong with the code that i can tell. 

Something about the delete method isn't being liked.

Anyone have any ideas?

DanrwAU, 2020-05-07:
```
::1 - - [07/May/2020:06:38:54 +0000] "DELETE /api/capcodes/104 HTTP/1.1" - - "http://localhost:3000/admin/aliases/104" 
```

Maybe i was wrong, it does seem to be submitted, but nothing happens to it.

SloCompTech, 2020-05-07:
Never used Angular 1 only 2+, but is here is missing *delete* option
https://github.com/pagermon/pagermon/blob/5aaf1f0e8655d66f3a522a4af5f97785a2bbc63d/server/themes/default/public/javascripts/admin/admin.main.js#L9
 ?

Called from here https://github.com/pagermon/pagermon/blob/5aaf1f0e8655d66f3a522a4af5f97785a2bbc63d/server/themes/default/public/javascripts/admin/admin.main.js#L462

DanrwAU, 2020-05-07:
The documentation i found seemed to suggest delete was an inherent option. Save is also missing but works... 

Either way i did try creating it there with no luck

DanrwAU, 2020-05-08:
Spent a few hrs messing around with this again last night, couldn't get it going. Code hasn't changed since original version, any ideas @davidmckenzie ?

davidmckenzie, 2020-05-08:
My prod's on 0.3.5 - deleting from list view works, deleting from detail page times out. It's definitely firing the request, just getting a timeout, so don't think anything wrong on frontend side, it's likely that API route is just broken.

It never hits `logger.main.info('Deleting '+id);` so it's either a problem with `router.delete` calls in general, or something weird with `var id = parseInt(req.params.id, 10);`

DanrwAU, 2020-05-08:
i think it's with the router.delete. 

I actually put a console log above that and it never gets triggered :(

PR #404 (2020-05-09): Fix delete button for Aliases
# Description

Fixes delete button not working on AliasDetail screen

Fixes #402 

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)

# How Has This Been Tested?

Delete capcodes, now working!

# Checklist:

- [x] My code follows the style guidelines of this project
- [x] I have performed a self-review of my own code
- [x] I have commented my code, particularly in hard-to-understand areas
- [x] I have updated changelog.md with changes made
closed 2020-05-11 by DanrwAU