pagermon/pagermon #410: Advance Message Filter

HobbyKonijn, 2020-05-16:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Possibillity to filter keywords in a message (example of a FLEX output)
16-05-2020 12:24:20: 000923118: A2 VWS N830 - Steenweg 4181AL Waardenburg 58835

Filtering on messages
Like top message have a filter at "N830" aka Roadnumber or "4181__" aka zipcode starting with 4181
All cases matched handeling should be equal as Alias, so you have a possibility to set Text description of the alias as "Incident at N830" and sent out a mail

**Describe alternatives you've considered**
Tekst Matches is setting menu, but this will only highlight a match.
If Tekst Matches can be set to sending e-mail or other broadcast would be also good.
16-05-2020 12:24:20: 000923118: A2 VWS N830 - Steenweg 4181AL Waardenburg 58835
I have a filter on Capcode: 000923118
But I'm not interested in 000923118: A2 (A2) aka prio setting ambu services but i'm interested in for example 000923118: A1 or like "A1 & N830" to have a notification in my mailbox

**Additional context**
000923118 can be showen at the dashboard for all type of Prio's locations and more.
But if any message contains like A1 or Grip & N830 or A1 & Waardenburg should trigger a mail

Maaaaattee, 2020-05-17:
Heya, have you had a look at the regex filter that already exists in the settings? Combine that with something like Pushover and you might be able to achieve what you are asking. :)

HobbyKonijn, 2020-05-17:
@Maaaaattee If I read correctly regex is only replacing certain strings in a message when comming in.
If I add in the regex "N830" to be replaced by "Highway" my incomming message will change.
Before:
16-05-2020 12:24:20: 000923118: A2 VWS **N830** - Steenweg 4181AL Waardenburg 58835
After:
16-05-2020 12:24:20: 000923118: A2 VWS **Highway** - Steenweg 4181AL Waardenburg 58835

This is not really what i'm looking for.  I looking for if parts of the message excists a email via SMTP will be sent out.

Maaaaattee, 2020-05-17:
I think you are thinking of the "Highlight" function. That does as you mentioned above by replacing certain words.
If you scroll further down there is a checkbox that is "Filter" you can add a regex to filter your messages by whatever you put in.... then, setup Pushover to send a notification for all those messages that it lets through.

Maaaaattee, 2020-05-17:
For example, I have in mine - If it contains the word "Purple" it allows that message in, then pushover sends me a notification.

HobbyKonijn, 2020-05-17:
Pushover and Prowl are external apps and my preferences is to handle it locally via my own mail server.  But to be honest I have no experiences in app setups like listed 2 ... but you made me think.  But if I add purple as keyword, does it send all messages out disregarding capcode?

DanrwAU, 2020-05-17:
I'd suggest heading over to discord.. Someone over there should be able to elaborate on how you can acheive this.

Maaaaattee, 2020-05-17:
What @DanrwAU said 😀

closed 2020-07-04 by DanrwAU