pagermon/pagermon #414: Discord Bug

JimmyRussells, 2020-05-29:
After you have setup discord. 
As per the Pagermon wiki to send messages to discord does not work.

**To Reproduce**
Make a Discord account.
Make a discord channel.
Then go to settings then to web hooks make a web hook with a name a channel to go to then copy the web hook.
Go to Pagermon in the setting put a tick in Discord then go to.
Aliases make a alias put in the cap code, agency, alias name then tick the box Discord and put in your web hook.

**Expected behavior**
Message failed to deliver. StatusCodeError: 500 - "{}"
Retrying in 1000 ms 

**Environment information:**
[pmon]  2020-05-29 07:39:01  [34mdebug[39m : ======================
[pmon]  2020-05-29 07:39:01  [34mdebug[39m : ======================
[pmon]  2020-05-29 07:39:01  [34mdebug[39m : plugin: Slack
[pmon]  2020-05-29 07:39:01  [34mdebug[39m : ======================
[pmon]  2020-05-29 07:39:01  [34mdebug[39m : plugin: Discord
[pmon]  2020-05-29 07:39:01  [34mdebug[39m : RUNNING PLUGIN!
[pmon]  2020-05-29 07:39:01  [31merror[39m : discord.RichEmbed is not a constructor
::ffff:192.168.1.145 - - [28/May/2020:21:39:01 +0000] "POST /api/messages HTTP/1.1" 500 2 "-" "PagerMon reader.js"
[pmon]  2020-05-29 07:39:02  [32minfo[39m : Ignoring duplicate: '19:39 P175666.MDA'
::ffff:192.168.1.145 - - [28/May/2020:21:39:02 +0000] "POST /api/messages HTTP/1.1" 200 18 "-" "PagerMon reader.js"
[pmon]  2020-05-29 07:39:06  [34mdebug[39m : beforeMessage start
[pmon]  2020-05-29 07:39:06  [34mdebug[39m : ======================
[pmon]  2020-05-29 07:39:06  [34mdebug[39m : trigger: message scope: before
[pmon]  2020-05-29 07:39:06  [34mdebug[39m : ======================
[pmon]  2020-05-29 07:39:06  [34mdebug[39m : data object
[pmon]  2020-05-29 07:39:06  [34mdebug[39m : {

**Additional context**
discordjs might have updated and needs to be looked at.

DanrwAU, 2020-06-03:
Source of the problem:
```
discord.RichEmbed is not a constructor
```

Investigated and found discord.js have implemented breaking changes.
Package.json has "latest" for discord.js.

Need to roll this back to an working version

marshyonline, 2020-06-03:
needs to be `"discord.js": "~11.6.4", `

PR #415 (2020-06-03): Fix Discord Plugin
# Description

Rollback Discord.js package to a version compatible with our implementation

Fixes #414 

## Type of change

Please delete options that are not relevant.

- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ] This change requires a documentation update

# How Has This Been Tested?

Rolled back discord.js package to 11.6.4 

Tested pushing messages to discord - see #pagermon-feed on disc [... cut]
closed 2020-06-03 by marshyonline