pagermon/pagermon #426: Group messages not working anymore on pagerpi image dus to new multimon version

woestric, 2020-06-26:
**Describe the bug**
The new pagermon image now has the new version of multimon-ng in it.
Unfortunatly it renders pagermon quite useless for dutch p2000 messages.
p2000 broadcasts group messages. Same message for different capcodes accompanied by a group capcode in the 2029568 to 20209575.
Multimon output would return each message seperatly. But the new version only sends 1 message with all corresponding capcodes in it. 
reader.js will only filter the first capcode, usually 2029568, and send that message through. All other capcodes are ignored.
So we miss most of the capcode messages.

**To Reproduce**
Steps to reproduce the behavior:
Use the new image and receive duth p2000 messages

**Expected behavior**
That pagermon can filter out the different capcodes and will send them all to the database.
Great feature would be that pagermon can show the message on one line adn the capcodes with aliases beneath that.
Something like 
date time, source, message
capcode 1, discipline, alias,
capcode 2, discipline, alias

**Screenshots**
Pagermon 3.5 with 'old' multimon version, reader.js output
[https://1drv.ms/u/s!AiK9Z_Y-pgVug5lk-QWklcI7okDZnw?e=gprgVi](url)
Pagermon 3.9 with new multimon, same timeframe
[https://1drv.ms/u/s!AiK9Z_Y-pgVug5ljDFG0mU_dw_EyVg?e=COPm6X](url)

**Environment information:**
 - OS: [e.g. Server on Ubuntu on Amazon EC2, Client on MacOS High Sierra]
 - Other details:

**Additional context**
I think this actually is a feature request, but still;-)
Example output multimon-ng raw message: FLEX|2020-05-14 16:47:07|1600/2/K/A|11.109|002029568 000108999 000104999 000104761 000104760 000104759 000104758 000104528 000103566 000103550|ALN|Herhaalde alarmering P 2 Herbez./kazern. 11-Kazerne Purmerend Gorslaan Purmerend

eopo, 2020-07-15:
Hey @woestric,
sounds interesting. I'll try to restore the previous behaviour for you.
Your feature idea of the message being shown on one line with all capcodes is a bit more complicated, since it needs changes in how the data is stored

woestric, 2020-07-15:
Hi @eopo ,

Thnx for picking it up.
I agree about the feature, I studied it myself and got some results. But the main problem is that by pushing each message in a group message sepratly, the build up on the webpage gets distorded. It looks like you have to store and push all the capcodes at once and probably with an extra field for identifying it as group message. But then you have the problem that you don't know with is the last capcode that has been sent....  It does sound like a great challange though :-)

eopo, 2020-07-15:
I'm thinking about eventually storing only unique messages and referencing them. If e.g. 100 capcodes get sent the same message (Regardless of the time), the message is only stored once, thus saving storage space.
Then, identical messages that were sent below a certain time delta threshold (Maybe user defined) could be shown as groups message.

woestric, 2020-07-17:
I think it will be a good approach and it might even be a necessary one, In the dutch P2000 system they will be sent within the same second (less). But because multiple emergency call centers can send message, sometimes another message slips through the groep messages. I don't know what ahppens in mutlimon-ng if that is the case. The first capcode in a group message is always in the 002029568 to 002029575 range.
Still I think just comparing the message within a short time periode will be sufficient. Question is when you want to submit/push the message. If you can push the message with all te correspondend codes, it will be easier to adjust the webpage. I think David previously looked into a sort of accordeon view..